mirsadm/motioncam #8: There's no support for CaptureResult ColorCorrectionTransform

eszdman, 2021-03-31:
On some devices(always undersaturated if wrongly choosed) color correction transform from capture result provides better color accuracy than one from CameraCharacteristics, suggest to add it in dev settings
Wrong matrices for example
[image]

mirsadm, 2021-03-31:
That's interesting. On my device it is not included in the capture result. I'll see if I can change it to prefer the capture result if it's there otherwise fallback camera one.

eszdman, 2021-03-31:
It's better to prefer most saturated one, for better compatibility

mirsadm, 2021-04-16:
This should be fixed now. Have not build a new APK yet though

closed 2021-04-16 by mirsadm