mirsadm/motioncam #9: Can your processing work on bayer data from non-android system?

f0enix, 2021-04-14:
Hi,
I have been doing some work on raw CFA data coming from raspberry pi having Sony camera sensors. I have used libraw/dcraw in the past to get previewable image from such data. I came across your repository and it seems that you are using Halide for real time demosaicing and image stacking for noise reduction which i find interesting.

I tried creating a metadata file containing information of the `RGGB` bayer along with raw bayer files and feed it to your algorithm but the output is messed up. It seems a channel/color issue. So my question is that whether your implementation is only specific to Android bayer data?

Here is my metadata file:
```json
{
    "colorIlluminant2": "standarda",
    "whiteLevel": 1095,
    "colorMatrix1": [1.506935715675354, -0.8880840539932251, -0.13350626826286316, -0.53555202484130859, 1.5421112775802612, 0.1474035233259201, -0.019240813329815865, 0.095311708748340607, 0.76038378477096558],
    "apertures": [1.6959937810599961],
    "colorMatrix2": [0.82994842529296875, -0.27388653159141541, -0.097103022038936615, -0.57290846109390259, 1.3635042905807495, 0.17090092599391937, -0.086338728666305542, 0.13636419177055359, 0.53642231225967407],
    "sensorArrangment": "rggb",
    "forwardMatrix2": [],
    "blackLevel": [64, 64, 64, 64],
    "focalLengths": [4],
    "colorIlluminant1": "d65",
    "writeDNG": true,
    "frames": [{
        "exposureTime": 333333333.33333331,
        "orientation": 0,
        "timestamp": "1598542181000000",
        "filename": "frame0.raw",
        "asShotNeutral": [0.49100935459136963, 1, 0.4330725371837616],
        "width": 4032,
        "rowStride": 4032,
        "height": 3024,
        "colorCorrectionGains": [0, 0, 0, 0],
        "lensShadingMapHeight": 0,
        "lensShadingMapWidth": 0,
        "iso": 800
    },{
        "exposureTime": 333333333.33333331,
        "orientation": 0,
        "timestamp": "1598542181000000",
        "filename": "frame1.raw",
        "asShotNeutral": [0.49100935459136963, 1, 0.4330725371837616],
        "width": 4032,
        "rowStride": 4032,
        "height": 3024,
        "colorCorrectionGains": [0, 0, 0, 0],
        "lensShadingMapHeight": 0,
        "lensShadingMapWidth": 0,
        "iso": 800
    },{
        "exposureTime": 333333333.33333331,
        "orientation": 0,
        "timestamp": "1598542181000000",
        "filename": "frame2.raw",
        "asShotNeutral": [0.49100935459136963, 1, 0.4330725371837616],
        "width": 4032,
        "rowStride": 4032,
        "height": 3024,
        "colorCorrectionGains": [0, 0, 0, 0],
        "lensShadingMapHeight": 0,
        "lensShadingMapWidth": 0,
        "iso": 800
    },{
        "exposureTime": 333333333.33333331,
        "orientation": 0,
        "timestamp": "1598542181000000",
        "filename": "frame3.raw",
        "asShotNeutral": [0.49100935459136963, 1, 0.4330725371837616],
        "width": 4032,
        "rowStride": 4032,
        "height": 3024,
        "colorCorrectionGains": [0, 0, 0, 0],
        "lensShadingMapHeight": 0,
        "lensShadingMapWidth": 0,
        "iso": 800
    },{
        "exposureTime": 333333333.33333331,
        "orientation": 0,
        "timestamp": "1598542181000000",
        "filename": "frame4.raw",
        "asShotNeutral": [0.49100935459136963, 1, 0.4330725371837616],
        "width": 4032,
        "rowStride": 4032,
        "height": 3024,
        "colorCorrectionGains": [0, 0, 0, 0],
        "lensShadingMapHeight": 0,
        "lensShadingMapWidth": 0,
        "iso": 800
    },{
        "exposureTime": 333333333.33333331,
        "orientation": 0,
        "timestamp": "1598542181000000",
        "filename": "frame5.raw",
        "asShotNeutral": [0.49100935459136963, 1, 0.4330725371837616],
        "width": 4032,
        "rowStride": 4032,
        "height": 3024,
        "colorCorrectionGains": [0, 0, 0, 0],
        "lensShadingMapHeight": 0,
        "lensShadingMapWidth": 0,
        "iso": 800
    },{
        "exposureTime": 333333333.33333331,
        "orientation": 0,
        "timestamp": "1598542181000000",
        "filename": "frame6.raw",
        "asShotNeutral": [0.49100935459136963, 1, 0.4330725371837616],
        "width": 4032,
        "rowStride": 4032,
        "height": 3024,
        "colorCorrectionGains": [0, 0, 0, 0],
        "lensShadingMapHeight": 0,
        "lensShadingMapWidth": 0,
        "iso": 800
    },{
        "exposureTime": 333333333.33333331,
        "orientation": 0,
        "timestamp": "1598542181000000",
        "filename": "frame7.raw",
        "asShotNeutral": [0.49100935459136963, 1, 0.4330725371837616],
        "width": 4032,
        "rowStride": 4032,
        "height": 3024,
        "colorCorrectionGains": [0, 0, 0, 0],
        "lensShadingMapHeight": 0,
        "lensShadingMapWidth": 0,
        "iso": 800
    }],
    "referenceTimestamp": "1598542181000000",
    "pixelFormat": "raw10",
    "postProcessingSettings": {
        
    },
    "forwardMatrix1": []
}
```

Note that i did not have information about the `rowStride` so i set it same as the width.
Since i don't have the lensShadingMap data, i set `lensShadingMapHeight` and `lensShadingMapWidth` to 0. This is what the final output image looks like:
[image]

Is the above json enough for the algorithm to work with or does it need additional metadata?

mirsadm, 2021-04-14:
It's hard to say. The metadata you've constructed is set to use RAW10 which is an Android specific format (https://developer.android.com/reference/android/graphics/ImageFormat#RAW10). Is it stored as 16 bits per pixel instead? If that's the case you can change the RAW10 to RAW16. The row stride is in bytes which should be set to width * (10 / 8) if using RAW10 or width * (16 / 8) if RAW 16.

mirsadm, 2021-04-14:
If you are able to send me the full file (frames + metadata you've constructed + some information about the camera) I can give it a go as well.

f0enix, 2021-04-14:
Thanks for the quick reply. Yes it is stored in 16 bits (i was not getting good results so i tried playing with the pixel format). Switching to `raw16` and using the correct `rowStride` improved the result but it is still pink
[image]

I currently do not have the camera on hand (due to it being in my office drawer and currently due to lockdown, don't know when i'll be able to bring it back). It is a small camera for embedded systems similar to the Sony IMX477 model but it allows capturing RGGB bayer. I had been capturing a lot of these raw data for analysis in the past and this is what i am currently using to experiment on libmotioncam.

Here is the link to the sample zip that i feed to the program:
[https://easyupload.io/rgtxt9](https://easyupload.io/rgtxt9)

mirsadm, 2021-04-15:
Hey @f0enix,

I had a quick look and there was a simple error in the metadata. The pixelFormat is stored per frame at the moment. Additionally, I've modified the black/white levels. It will probably need additional tweaking, here's the output:

[image]

And the metadata:

```
{
  "colorIlluminant2": "standarda",
  "whiteLevel": 16383,
  "colorMatrix1": [
    1.506935715675354,
    -0.8880840539932251,
    -0.13350626826286316,
    -0.53555202484130859,
    1.5421112775802612,
    0.1474035233259201,
    -0.019240813329815865,
    0.095311708748340607,
    0.76038378477096558
  ],
  "apertures": [
    1.6959937810599961
  ],
  "colorMatrix2": [
    0.82994842529296875,
    -0.27388653159141541,
    -0.097103022038936615,
    -0.57290846109390259,
    1.3635042905807495,
    0.17090092599391937,
    -0.086338728666305542,
    0.13636419177055359,
    0.53642231225967407
  ],
  "sensorArrangment": "bggr",
  "forwardMatrix2": [],
  "blackLevel": [
    2127,
    2127,
    2127,
    2127
  ],
  "focalLengths": [
    4
  ],
  "colorIlluminant1": "d65",
  "writeDNG": true,
  "frames": [
    {
      "exposureTime": 333333333.33333331,
      "orientation": 0,
      "timestamp": "1598542181000000",
      "filename": "frame0.raw",
      "asShotNeutral": [
        0.49100935459136963,
        1,
        0.4330725371837616
      ],
      "width": 4032,
      "rowStride": 8064,
      "height": 3024,
      "colorCorrectionGains": [
        0,
        0,
        0,
        0
      ],
      "lensShadingMapHeight": 0,
      "lensShadingMapWidth": 0,
      "iso": 800,
      "pixelFormat": "raw16"
    },
    {
      "exposureTime": 333333333.33333331,
      "orientation": 0,
      "timestamp": "1598542181000000",
      "filename": "frame1.raw",
      "asShotNeutral": [
        0.49100935459136963,
        1,
        0.4330725371837616
      ],
      "width": 4032,
      "rowStride": 8064,
      "height": 3024,
      "colorCorrectionGains": [
        0,
        0,
        0,
        0
      ],
      "lensShadingMapHeight": 0,
      "lensShadingMapWidth": 0,
      "iso": 800,
      "pixelFormat": "raw16"
    },
    {
      "exposureTime": 333333333.33333331,
      "orientation": 0,
      "timestamp": "1598542181000000",
      "filename": "frame2.raw",
      "asShotNeutral": [
        0.49100935459136963,
        1,
        0.4330725371837616
      ],
      "width": 4032,
      "rowStride": 8064,
      "height": 3024,
      "colorCorrectionGains": [
        0,
        0,
        0,
        0
      ],
      "lensShadingMapHeight": 0,
      "lensShadingMapWidth": 0,
      "iso": 800,
      "pixelFormat": "raw16"
    },
    {
      "exposureTime": 333333333.33333331,
      "orientation": 0,
      "timestamp": "1598542181000000",
      "filename": "frame3.raw",
      "asShotNeutral": [
        0.49100935459136963,
        1,
        0.4330725371837616
      ],
      "width": 4032,
      "rowStride": 8064,
      "height": 3024,
      "colorCorrectionGains": [
        0,
        0,
        0,
        0
      ],
      "lensShadingMapHeight": 0,
      "lensShadingMapWidth": 0,
      "iso": 800,
      "pixelFormat": "raw16"
    },
    {
      "exposureTime": 333333333.33333331,
      "orientation": 0,
      "timestamp": "1598542181000000",
      "filename": "frame4.raw",
      "asShotNeutral": [
        0.49100935459136963,
        1,
        0.4330725371837616
      ],
      "width": 4032,
      "rowStride": 8064,
      "height": 3024,
      "colorCorrectionGains": [
        0,
        0,
        0,
        0
      ],
      "lensShadingMapHeight": 0,
      "lensShadingMapWidth": 0,
      "iso": 800,
      "pixelFormat": "raw16"
    },
    {
      "exposureTime": 333333333.33333331,
      "orientation": 0,
      "timestamp": "1598542181000000",
      "filename": "frame5.raw",
      "asShotNeutral": [
        0.49100935459136963,
        1,
        0.4330725371837616
      ],
      "width": 4032,
      "rowStride": 8064,
      "height": 3024,
      "colorCorrectionGains": [
        0,
        0,
        0,
        0
      ],
      "lensShadingMapHeight": 0,
      "lensShadingMapWidth": 0,
      "iso": 800,
      "pixelFormat": "raw16"
    },
    {
      "exposureTime": 333333333.33333331,
      "orientation": 0,
      "timestamp": "1598542181000000",
      "filename": "frame6.raw",
      "asShotNeutral": [
        0.49100935459136963,
        1,
        0.4330725371837616
      ],
      "width": 4032,
      "rowStride": 8064,
      "height": 3024,
      "colorCorrectionGains": [
        0,
        0,
        0,
        0
      ],
      "lensShadingMapHeight": 0,
      "lensShadingMapWidth": 0,
      "iso": 800,
      "pixelFormat": "raw16"
    },
    {
      "exposureTime": 333333333.33333331,
      "orientation": 0,
      "timestamp": "1598542181000000",
      "filename": "frame7.raw",
      "asShotNeutral": [
        0.49100935459136963,
        1,
        0.4330725371837616
      ],
      "width": 4032,
      "rowStride": 8064,
      "height": 3024,
      "colorCorrectionGains": [
        0,
        0,
        0,
        0
      ],
      "lensShadingMapHeight": 0,
      "lensShadingMapWidth": 0,
      "iso": 800,
      "pixelFormat": "raw16"
    }
  ],
  "referenceTimestamp": "1598542181000000",
  "pixelFormat": "raw16",
  "postProcessingSettings": {},
  "forwardMatrix1": []
}
```

f0enix, 2021-04-15:
> The pixelFormat is stored per frame at the moment

I was assuming that it would be stored at global level since all frames are most likely to have same pixel format.

I tried your modified metadata and on mine the results look slight different with a green tint instead.
[image]

I am currently using the latest version on master. 

Furthermore i found that the dng generated is corrupted and cannot be opened in Photoshop/Lightroom but is working on RawDigger/RawTherapee. Not sure if you are aware of it but just wanted to point it out.
[image]

mirsadm, 2021-04-15:
In the metadata I posted above I had changed the sensorArrangement to BGGR as a test. You'd want to change that back to RGGB.

I don't have Lightroom to test with on my Mac but I am able to open the DNGs generated by the Android app in the mobile version of Lightroom. It may be because of the missing lens shading map or something along those lines that it's failing to open that particular DNG.

Is the image I posted correct? I can't tell if the colours are accurate.

f0enix, 2021-04-15:
I changed to it RGGB and the result is much better. If i open the dng in mac quick preview, it looks close to the original photo but the jpg version has strong red tint. Here is the comparison:
[image]

[image]

I don't have an exact rgb version of that raw bayer because i had been applying some post processing settings on it. But to give you an idea, here is how it looked like with those settings:
[image]

Maybe the red tint has something to do with the XYZ mapping from the camera colorspace to the srgb colorspace or is it a white balance issue?

mirsadm, 2021-04-15:
You are right. The missing forward matrix was causing it to miscalculate the sRGB matrix. I've pushed a fix. Here's the output now:

[image]

f0enix, 2021-04-15:
Now it looks good.

Btw i just tried to convert a raw frame to srgb using the `createPreview` method. I used a downscaling of 2 and was expecting it to result in an image of `1512x2016` px but instead the result was `756x1008` px. It seems that it is apply a double downscaling. Is this the expected behaviour?

```c++
auto frame = rawContainer.loadFrame(*it);
Halide::Runtime::Buffer<uint8_t>framePreview = createPreview(*frame, 2, rawContainer.getCameraMetadata(),rawContainer.getPostProcessSettings());
cv::Mat framePreviewImg(framePreview.height(),framePreview.width(),CV_8UC4,framePreview.data());
cv::cvtColor(framePreviewImg, framePreviewImg, cv::COLOR_BGRA2RGBA);
cv::imwrite("preview.jp", framePreviewImg);
```

mirsadm, 2021-04-15:
Yep, I use that mostly to create a fast preview in the Android app. It is a rough estimate of the full post processing. The reason is that the preview code is simply averaging the green channels and not demosaicing to a full sized image. So without downscaling it'll already be half the width and height.

f0enix, 2021-04-16:
ok thanks

closed 2021-04-16 by f0enix