mirsadm/motioncam #12: App crashes reading corrupted zip file

paolodepetrillo, 2021-04-17:
I was experimenting with the app and it created a corrupt zip file under files/motionCam. Not sure what happened, maybe it exited before it finished writing the file. When the app is relaunched, it crashes when it tries to process the file because ZipReader throws a C++ exception which is not caught.
This patch fixes the problem by catching the C++ exception and re-throwing it as a Java RuntimeException. I've never worked with JNI so I'm not really sure if this is the right way to fix it, but it seems to work.

[process_crash.patch.txt](https://github.com/mirsadm/motioncam/files/6328589/process_crash.patch.txt)

When the app was relaunched with this fix, it logged the exception and then ran normally:
`2021-04-16 23:32:05.343 20282-20377/com.motioncam W/System.err: java.lang.RuntimeException: Can't read /data/user/0/com.motioncam/files/motionCam/IMG_20210416_230301.zip
2021-04-16 23:32:05.344 20282-20377/com.motioncam W/System.err:     at com.motioncam.processor.NativeProcessor.ProcessFile(Native Method)
2021-04-16 23:32:05.344 20282-20377/com.motioncam W/System.err:     at com.motioncam.processor.NativeProcessor.processFile(NativeProcessor.java:25)
2021-04-16 23:32:05.344 20282-20377/com.motioncam W/System.err:     at com.motioncam.processor.ProcessorService$ProcessFile.call(ProcessorService.java:91)`

mirsadm, 2021-04-17:
Cheers, I'll take a look at this soon. The error handling definitely needs more work.

mirsadm, 2021-04-23:
Fixed, thanks for that.

closed 2021-04-23 by mirsadm