mirsadm/motioncam #13: Pixel 4a OpenCL support

paolodepetrillo, 2021-04-17:
OpenCL does not work on the Pixel 4a (and probably other Pixels), because /vendor/lib64/libOpenCL.so is not listed in /vendor/etc/public.libraries.txt, so user applications are not allowed to dlopen it.

However there is an undocumented workaround, there is a wrapper library /vendor/lib64/libOpenCL-pixel.so, which can be used by an application. This workaround is used by [TensorFlow opencl_wrapper.cc](https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/delegates/gpu/cl/opencl_wrapper.cc).

I don't know if this is really a stable, fully functional implementation or if it might stop working in the future. But I did test it and it seems to work with the dual exposure preview.

As a quick test I modified halide's src/runtime/opencl.cc. I'm not sure if there is any cleaner way to do this without having to maintain a change to halide source.

[pixelopencl.patch.txt](https://github.com/mirsadm/motioncam/files/6329429/pixelopencl.patch.txt)

mirsadm, 2021-04-18:
Thanks, that is a good find. I did receive crash reports from Pixel owners. I figured Google has disabled OpenCL for some odd reason. It might be worthwhile submitting this patch to Halide as well

mirsadm, 2021-04-19:
I've applied this patch to my fork of Halide. Thanks for the patch!

closed 2021-04-19 by mirsadm

locuturus, 2021-05-22:
For what it's worth Google is moving towards actually supporting opencl (finally) rather than renderscript in Android 12:  [XDA Article](https://www.xda-developers.com/renderscript-deprecated-android-12/)

paolodepetrillo, 2021-05-22:
It looks like they are recommending Vulkan, not OpenCL, to replace Renderscript.

mirsadm, 2021-05-24:
Renderscript has been a bit of a disaster IMO so I'm glad they're moving away from it. Unfortunately I don't think Vulkan compute shaders are in good position at the moment either. OpenCL still seems to be the way to go even though its support is somewhat limited.