mirsadm/motioncam #14: Enabling dual exposure controls either force closes the app or makes it not function on Pixel 3

just-a-user-sudo, 2021-04-23:
Disabling dual exposure controls lets the app work as normal, and I am able to take photos, edit them, and export them as normal, similar to the experience of the app before dual exposure controls were introduced.

Enabling them, however, makes the app close before anything shows up in the viewfinder. The UI is visible for a moment.

https://user-images.githubusercontent.com/80051348/115934730-42992000-a481-11eb-9064-c68512da9987.mp4

mirsadm, 2021-04-24:
Can you try installing a newer version from:
https://github.com/mirsadm/motioncam/releases

That issue should be fixed. You might need to kill the app and restart it if the viewfinder is blank after first enabling dual exposure. Working on a fix for that.

just-a-user-sudo, 2021-04-24:
Ah, my fault for not checking for a new version. You are correct, it has been fixed.

closed 2021-04-24 by just-a-user-sudo