mirsadm/motioncam #15: Usability Suggestion

locuturus, 2021-04-27:
Hello, I have 2 quick suggestions for ease of use:

* Shortcut to open an external gallery app.
* Allow volume button to work the shutter.

I'm loving the concept you have here!

mirsadm, 2021-04-27:
- Allow volume button to work the shutter.
Have done that one.

- Shortcut to open an external gallery app.
I've been meaning to do this for a long time. Will see if I can get around to it soon-ish.

locuturus, 2021-05-11:
* Allow volume button to work the shutter.
Have done that one.

In pre release? It's not working for me in 2.8.2 but maybe it's not in there yet.

mirsadm, 2021-05-11:
HI,

Yep I've made a new release which has volume down to capture, volume up for burst and also provides a preview of the capture and shortcuts to share/open in gallery.

You can grab it here:
https://github.com/mirsadm/motioncam/releases/download/3.0.1/MotionCam.3.0.1.apk

Cheers

closed 2021-05-11 by mirsadm