mirsadm/motioncam #16: Building denoise generator fails when enabling GPU features

l0stpenguin, 2021-05-02:
Hi,

I just came across this project and tried it on my mac using the xcode project. It worked well when i provided it with a zip file of raw data. I wanted to compile it using metal so that it uses gpu on my mac to see how fast it can go. 
So i edited the `generate.sh` script to use metal as features:

`build_denoise  x86-64-osx-metal  mac-metal` 

But this fails when building:
```
[mac-metal] Building forward_transform_generator
+ ./tmp/denoise_generator -g forward_transform_generator -f forward_transform -e static_library,h -o ../halide/mac-metal target=x86-64-osx-metal-no_runtime input.type=uint16 levels=6
Unhandled exception: Internal Error at /workspace/motioncam/tmp/halide-src/src/CodeGen_Metal_Dev.cpp:209 triggered by user code at : Condition failed: !function_takes_user_context(op->name): 
```

i also tried with `build_denoise  x86-64-osx-metal-user_context  mac-metal` but it produced same error.

Does it mean that the denoise generator cannot run on GPU? Or it is possible that i might be compiling it wrongly?

mirsadm, 2021-05-03:
It can run on the GPU. The Halide schedule is out of date for the wavelet transforms and I haven't written one for the DenoiseGenerator. My laptop has a pretty old integrated GPU so it doesn't run any better on it. With a more modern/dedicated GPU it would run a lot faster.

l0stpenguin, 2021-05-03:
On the README, it says 

> Motion Cam uses the GPU to generate a real time preview of the camera from its RAW data

So if understand correctly, on Android only the preview generation uses GPU but not the denoising operation?

What would be required if i want to modify the denoise generator to be ran on GPU?

mirsadm, 2021-05-03:
Yeah just the preview runs on the GPU. FYI I did experiment with running the denoising on the GPU on Android and it wasn't any faster than on the CPU. The other issue is that it causes the UI to become very sluggish (even after leaving the app) as the GPU becomes heavily utilised. IMO if you're after speeding up the denoising I would investigate getting it to run on the Hexagon DSP instead. It would use significantly less power. If you still want to give it a go have a look at the schedule_for_gpu() functions as a starting point (and Halide tutorial as well https://halide-lang.org/tutorials/tutorial_introduction.html).

closed 2021-05-03 by l0stpenguin

l0stpenguin, 2021-05-09:
How did you do this?
> I did experiment with running the denoising on the GPU on Android and it wasn't any faster than on the CPU.

I have configuring the schedular manually using the schedule_for_gpu() but i am not good at figuring the optimised way of doing it. So i used the `Li2018` autoscheduler to generate a starting point for the gpu schedule. 

```
./tmp/denoise_generator -g denoise_generator -f fuse_denoise  -e schedule,static_library,h -o ../halide/${ARCH} target=${TARGET}-${FLAGS} 
input0.type=uint16 input1.type=uint16 pendingOutput.type=float32 output.type=float32 -p ${HALIDE_PATH}/lib/libautoschedule_li2018.so -s Li2018 auto_schedule=true
```
Unfortutnately that also failed with this error:
```
Unhandled exception: Error: Signed integer overflow occurred during constant-folding. Signed integer overflow for int32 and int64 is undefined behavior in Halide.
```
Do you have a basic gpu schedule i can use as starting point for the gpu schedule?