mirsadm/motioncam #18: Lucky Features

eszdman, 2021-05-26:
-Use gyroscope to select best frame as default
-Use gyroscope and sensor noise model to choose best iso/exposure time, automatics isn't designed for stacking

mirsadm, 2021-05-27:
-Use gyroscope to select best frame as default

In burst mode it will attempt to measure the sharpness of the first few images and pick the best one. I'm not sure how the gyro would help there? I did use to do the same in normal photo mode but took it out because it can lead to images being captured that were too far from the time the shutter button was pushed. I might put that back in but instead only choose from 1 or 2 images closest to the time the shutter button was pressed.

 -Use gyroscope and sensor noise model to choose best iso/exposure time, automatics isn't designed for stacking

If you shoot in "night mode" it does override the iso/shutter speed. It behaves in a similar way to enhanced HDR+ in GCam. I did attempt to reimplment the auto exposure to pick more approrpiate settings but it is difficult to make it work quickly enough. The feedback loop between measuring and adjusting the settings is too long. On my phone the automatic exposure tends to pick much faster shutter speeds than the built in camera.

eszdman, 2021-05-27:
> it is difficult to make it work quickly enough
PID controllers are used for such systems
> tends to pick much faster shutter speeds
You need to use SENSOR_NOISE_PROFILE from captureResult, to choose the best possible iso and exposure
> I'm not sure how the gyro would help there?
And then decrease the exposure time if the gyroscope has too large PSF

closed 2021-11-03 by mirsadm