mirsadm/motioncam #21: Latest version causes purple color in overexposed areas

f0enix, 2021-07-30:
Hi,
I have some raw bayer data which i average using _libMotionCam_ binary. It was working fine but today i built the latest version from master branch (commit hash _83a52b7984d58ac8dc895ce02e42f63e31a50975_) and running it on same data, i noticed that all overexposed areas are being displayed as purple.
Here is a preview generated from such raw data:
[image]

Now if i average it by feeding it as a zip (containing all required metadata) to _libMotionCam_ binary on my mac, this is what i get:
[image]

i can deduce that the preview functions are working fine, it seems that the post processing is broken and some clipping is occurring in those bright regions. 
Note that same zip which contains multiple frames of this shot with same settings produces good output with earlier versions of libMotionCam binary. I tried with several images and all have this purple color in overexposed regions.

Is this a bug or does the latest version requires shooting dark exposures and needs HDR fusion to fix it?

mirsadm, 2021-07-30:
are you able to provide the zip file for me? it certinaly looks like a clipping bug

f0enix, 2021-07-30:
Here is the link to the zip:
[https://www.filemail.com/d/impgbqiqtdkdjfn](https://www.filemail.com/d/impgbqiqtdkdjfn)

mirsadm, 2021-07-30:
Cheers, I've downloaded it. Will look at it tonight if I can. Feel free to delete it from there.

mirsadm, 2021-08-03:
That's been fixed in the latest commit. Thanks for reporting that.

closed 2021-08-03 by mirsadm