mirsadm/motioncam #22: "Enable dual exposure controls" causes crash at startup

genpfault, 2021-08-04:
Checked "Enable dual exposure controls" and now the app crashes every time shortly after startup:

```
...
08-04 14:42:16.538 12455 12473 D ACameraDevice: Device error received, code 3, frame number 12, request ID 2, subseq ID 0
08-04 14:42:16.538   843 12493 E mm-camera: <MCT   ><ERROR> 1355: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr NULL
08-04 14:42:17.040   843 12493 I chatty  : uid=1047(cameraserver) CAM_MctBus_3 identical 15 lines
08-04 14:42:17.073   843 12493 E mm-camera: <MCT   ><ERROR> 1355: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr NULL
08-04 14:42:17.096 12455 12484 I halide  : Error: CL: clEnqueueNDRangeKernel failed: CL_INVALID_WORK_GROUP_SIZE
--------- beginning of crash
08-04 14:42:17.097 12455 12484 F libc    : Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 12484 (com.motioncam), pid 12455 (com.motioncam)
08-04 14:42:17.107   843 12493 E mm-camera: <MCT   ><ERROR> 1355: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr NULL
08-04 14:42:17.140   843 12493 I chatty  : uid=1047(cameraserver) CAM_MctBus_3 identical 1 line
08-04 14:42:17.175   843 12493 E mm-camera: <MCT   ><ERROR> 1355: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr NULL
08-04 14:42:17.193 12585 12585 I crash_dump64: obtaining output fd from tombstoned, type: kDebuggerdTombstone
08-04 14:42:17.194   743   743 I tombstoned: received crash request for pid 12484
08-04 14:42:17.197 12585 12585 I crash_dump64: performing dump of process 12455 (target tid = 12484)
08-04 14:42:17.210   843 12493 E mm-camera: <MCT   ><ERROR> 1355: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr NULL
08-04 14:42:17.219 12585 12585 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
08-04 14:42:17.219 12585 12585 F DEBUG   : LineageOS Version: '18.1-20210706-NIGHTLY-payton'
08-04 14:42:17.219 12585 12585 F DEBUG   : Build fingerprint: 'motorola/payton/payton:8.0.0/OPWS27.57-25-6-10/12:user/release-keys'
08-04 14:42:17.219 12585 12585 F DEBUG   : Revision: 'p4'
08-04 14:42:17.219 12585 12585 F DEBUG   : ABI: 'arm64'
08-04 14:42:17.220 12585 12585 F DEBUG   : Timestamp: 2021-08-04 14:42:17-0500
08-04 14:42:17.220 12585 12585 F DEBUG   : pid: 12455, tid: 12484, name: com.motioncam  >>> com.motioncam <<<
08-04 14:42:17.220 12585 12585 F DEBUG   : uid: 10243
08-04 14:42:17.221 12585 12585 F DEBUG   : signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
08-04 14:42:17.221 12585 12585 F DEBUG   :     x0  0000000000000000  x1  00000000000030c4  x2  0000000000000006  x3  0000006f83278f20
08-04 14:42:17.221 12585 12585 F DEBUG   :     x4  fefeff7184118d7f  x5  fefeff7184118d7f  x6  fefeff7184118d7f  x7  7f7f7f7fff7fffff
08-04 14:42:17.221 12585 12585 F DEBUG   :     x8  00000000000000f0  x9  caad997716eb8ac4  x10 0000000000000000  x11 ffffffc0fffffbdf
08-04 14:42:17.221 12585 12585 F DEBUG   :     x12 0000000000000001  x13 0000000000000046  x14 000577c4668b53e2  x15 00000000037a3c10
08-04 14:42:17.221 12585 12585 F DEBUG   :     x16 0000007285115c80  x17 00000072850f7c10  x18 0000006f8327a88c  x19 00000000000030a7
08-04 14:42:17.221 12585 12585 F DEBUG   :     x20 00000000000030c4  x21 00000000ffffffff  x22 00000070f23bd103  x23 0000006f88a5c180
08-04 14:42:17.221 12585 12585 F DEBUG   :     x24 0000000000000000  x25 0000006f8327a93c  x26 0000006f88a5c888  x27 0000000000000000
08-04 14:42:17.221 12585 12585 F DEBUG   :     x28 0000007072455670  x29 0000006f83278fa0
08-04 14:42:17.221 12585 12585 F DEBUG   :     lr  00000072850aa2a0  sp  0000006f83278f00  pc  00000072850aa2cc  pst 0000000000000000
08-04 14:42:17.223 12585 12585 F DEBUG   : backtrace:
08-04 14:42:17.223 12585 12585 F DEBUG   :       #00 pc 000000000004e2cc  /apex/com.android.runtime/lib64/bionic/libc.so (abort+164) (BuildId: be9c72fe4db37cd191b589b74d090d13)
08-04 14:42:17.223 12585 12585 F DEBUG   :       #01 pc 000000000058b36c  /data/app/~~f_aTfxMKP1HwvHHgHc6_3Q==/com.motioncam-o78GUdbPP0ZAzZkiQT-qNw==/base.apk!libnative-camera-opencl.so (offset 0x123b000) (halide_default_error+116) (BuildId: ac2b8a963bf02795076ba79c7e759ab878db889f)
...
```

Settings are otherwise default.

System info:

* Moto X4 (`payton`) running LineageOS 18.1 (July 6, 2021 build)
* Using MotionCam 4.0.0 from Github

mirsadm, 2021-08-05:
Cheers will take a look

genpfault, 2021-08-09:
Same behavior on a Moto G7 Play (`channel`) on LineageOS 18.1.

Different OpenCL error though:

```
...
08-09 13:21:47.212  8778  8778 I MotionCam: Focus state: ACTIVE_SCAN
08-09 13:21:47.212  8778  8807 D libMotionCam: AUTO_FOCUS_WAIT completed
08-09 13:21:47.213  8778  8807 D libMotionCam: setState(AUTO_FOCUS_WAIT -> TRIGGER_AUTO_FOCUS)
08-09 13:21:47.213  8778  8807 D libMotionCam: nextState() [mCaptureSessionState == CameraCaptureSessionState::READY]
08-09 13:21:47.213  8778  8807 D libMotionCam: triggerAutoFocus()
08-09 13:21:47.213  8498  8598 D EventBus: No subscribers registered for event class org.thoughtcrime.securesms.backup.FullBackupBase$BackupEvent
08-09 13:21:47.213  8498  8598 D EventBus: No subscribers registered for event class org.greenrobot.eventbus.NoSubscriberEvent
08-09 13:21:47.213  8778  8807 D libMotionCam: Camera session has changed state (1)
08-09 13:21:47.214  8778  8807 I MotionCam: Camera state changed ACTIVE
08-09 13:21:47.217  8778  8857 W libMotionCam: Out of buffers
08-09 13:21:47.220   640   640 I Zygote  : Process 8560 exited due to signal 9 (Killed)
08-09 13:21:47.221  1402  7391 I ActivityManager: Process com.google.process.gservices (pid 8560) has died: cch+15 CEM 
08-09 13:21:47.223  1402  1431 I libprocessgroup: Successfully killed process cgroup uid 10096 pid 8560 in 2ms
08-09 13:21:47.228  8498  8598 D EventBus: No subscribers registered for event class org.thoughtcrime.securesms.backup.FullBackupBase$BackupEvent
08-09 13:21:47.228  8498  8598 D EventBus: No subscribers registered for event class org.greenrobot.eventbus.NoSubscriberEvent
08-09 13:21:47.229  8778  8882 W Adreno-GSL: <os_lib_map:1435>:   os_lib_map error: dlopen failed: library "libllvm-qcom.so" not found, on 'libllvm-qcom.so'
08-09 13:21:47.229  8778  8882 W Adreno-CB: <cl_osal_open_library:326>: Unable to open library: error: (null), on 'libllvm-qcom.so'
08-09 13:21:47.232  8498  8598 D EventBus: No subscribers registered for event class org.thoughtcrime.securesms.backup.FullBackupBase$BackupEvent
08-09 13:21:47.232  8498  8598 D EventBus: No subscribers registered for event class org.greenrobot.eventbus.NoSubscriberEvent
08-09 13:21:47.235  8778  8882 W Adreno-GSL: <os_lib_map:1435>:   os_lib_map error: dlopen failed: library "libllvm-qcom.so" not found, on 'libllvm-qcom.so'
08-09 13:21:47.236  8778  8882 W Adreno-CB: <cl_osal_open_library:326>: Unable to open library: error: (null), on 'libllvm-qcom.so'
08-09 13:21:47.237  8778  8882 I halide  : Error: CL: clCreateContext failed: CL_OUT_OF_HOST_MEMORY:-6
--------- beginning of crash
08-09 13:21:47.237  8778  8882 F libc    : Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 8882 (com.motioncam), pid 8778 (com.motioncam)
08-09 13:21:47.243  8498  8598 D EventBus: No subscribers registered for event class org.thoughtcrime.securesms.backup.FullBackupBase$BackupEvent
08-09 13:21:47.243  8498  8598 D EventBus: No subscribers registered for event class org.greenrobot.eventbus.NoSubscriberEvent
08-09 13:21:47.246  8498  8598 D EventBus: No subscribers registered for event class org.thoughtcrime.securesms.backup.FullBackupBase$BackupEvent
08-09 13:21:47.246  8498  8598 D EventBus: No subscribers registered for event class org.greenrobot.eventbus.NoSubscriberEvent
08-09 13:21:47.252   737  8868 I mm-camera: <IMGLIB>< INFO> 3997: module_faceproc_client_print_config: FD CONFIG : engine=FACEPROC_ENGINE_SW, mode=1, enable_bsgc=0, chromatix id=20, chromatix used = fd_chromatix_detect_qc_moto_channel.h sw_detect=0, enable_upfront=1, sw_detect_enable_upfront=0
08-09 13:21:47.252   737  8868 I mm-camera: <IMGLIB>< INFO> 4012: module_faceproc_client_print_config: FD CONFIG : min_face_size=45, min_face_size_ratio=0.100000, speed=0, max_fps=15 (0 means max possible), thresholds= 770, 0 0 0, 0 0 0, search_cycles=10 10
08-09 13:21:47.252   737  8868 I mm-camera: <IMGLIB>< INFO> 4017: module_faceproc_client_print_config: FD CONFIG : Frame width=640, height=480, stride=640, scanline=480
08-09 13:21:47.253  8498  8598 D EventBus: No subscribers registered for event class org.thoughtcrime.securesms.backup.FullBackupBase$BackupEvent
08-09 13:21:47.253  8498  8598 D EventBus: No subscribers registered for event class org.greenrobot.eventbus.NoSubscriberEvent
08-09 13:21:47.256   737  8815 E libc    : Access denied finding property "gsm.operator.iso-country"
08-09 13:21:47.258  8498  8598 D EventBus: No subscribers registered for event class org.thoughtcrime.securesms.backup.FullBackupBase$BackupEvent
08-09 13:21:47.259  8498  8598 D EventBus: No subscribers registered for event class org.greenrobot.eventbus.NoSubscriberEvent
08-09 13:21:47.278  8885  8885 E DEBUG   : failed to readlink /proc/8882/fd/65: No such file or directory
08-09 13:21:47.279  8885  8885 E DEBUG   : failed to readlink /proc/8882/fd/82: No such file or directory
08-09 13:21:47.280  8498  8598 D EventBus: No subscribers registered for event class org.thoughtcrime.securesms.backup.FullBackupBase$BackupEvent
08-09 13:21:47.280  8498  8598 D EventBus: No subscribers registered for event class org.greenrobot.eventbus.NoSubscriberEvent
08-09 13:21:47.283  8498  8598 D EventBus: No subscribers registered for event class org.thoughtcrime.securesms.backup.FullBackupBase$BackupEvent
08-09 13:21:47.283  8498  8598 D EventBus: No subscribers registered for event class org.greenrobot.eventbus.NoSubscriberEvent
[... cut ...]
08-09 13:21:47.321  8885  8885 F DEBUG   : backtrace:
08-09 13:21:47.321  8885  8885 F DEBUG   :       #00 pc 0000000000089acc  /apex/com.android.runtime/lib64/bionic/libc.so (abort+164) (BuildId: 48a42e30f1ee71f42e5c1d9374f3ffc1)
08-09 13:21:47.322  8885  8885 F DEBUG   :       #01 pc 000000000058b36c  /data/app/~~4b4ElK2x8I6iTsdH2F4PYA==/com.motioncam-Q9B3QrbL3paN0YWFIjoPJw==/base.apk!libnative-camera-opencl.so (offset 0x123b000) (halide_default_error+116) (BuildId: ac2b8a963bf02795076ba79c7e759ab878db889f)
...
```

mirsadm, 2021-10-16:
Hey. Yeah it would be useful to get the logs if possible.

genpfault, 2021-10-20:
Same behavior in 5.0.3:

```
$ adb logcat --clear && adb logcat > log.txt
```

```
...
10-20 11:19:44.731   733  9655 I mm-camera: <ISP   >< INFO> 1427: isp_handler_module_handle_reg_update:  Warning! Invalid reg_update state 0
10-20 11:19:44.732   733  9581 E libc    : Access denied finding property "gsm.operator.iso-country"
10-20 11:19:44.729   733   733 W CAM_MctBus_1: type=1400 audit(0.0:1224): avc: denied { read } for uid=1047 name="u:object_r:radio_prop:s0" dev="tmpfs" ino=16388 scontext=u:r:hal_camera_default:s0 tcontext=u:object_r:radio_prop:s0 tclass=file permissive=0
10-20 11:19:44.765   733  9581 E mm-camera: <IFACE ><ERROR> 975: iface_axi_stream_request_frame: iface_axi_stream_request_frame: cannot find stream, session_id = 1, stream_id = 4
10-20 11:19:44.817   733  9596 E mm-camera: <STATS_AF ><ERROR> 4705: af_port_handle_pdaf_stats: Fail to init buf divert ack ctrl
10-20 11:19:44.835   733  9591 I mm-camera: <ISP   >< INFO> 230: ihist_stats44_stats_config_validate: warning: Invalid IHIST ROI from 3A 0 0 0 0
10-20 11:19:44.836   733  9591 I mm-camera: <ISP   >< INFO> 230: ihist_stats44_stats_config_validate: warning: Invalid IHIST ROI from 3A 0 0 0 0
10-20 11:19:44.842  9545  9563 E ACameraDevice: onDeviceIdle sending state cb
10-20 11:19:44.842  9545  9573 D libMotionCam: Camera session has changed state (0)
10-20 11:19:44.844   549   761 E SurfaceFlinger: setDesiredDisplayConfigSpecs: failed to call setDesiredDisplayConfigSpecs: -22
10-20 11:19:44.844  1427  1447 E SurfaceFlinger: setDesiredDisplayConfigSpecs failed to transact: -22
10-20 11:19:44.846  9545  9573 I MotionCam: Camera state changed READY
10-20 11:19:44.847  9545  9677 I libMotionCam: Setting up buffers
10-20 11:19:44.847  9545  9575 E ACameraMetadata: getConstEntry: cannot find metadata tag 917509
10-20 11:19:44.847  9545  9575 E ACameraMetadata: getConstEntry: cannot find metadata tag 917510
10-20 11:19:44.847  9545  9575 E ACameraMetadata: getConstEntry: cannot find metadata tag 917513
10-20 11:19:44.847  9545  9575 E ACameraMetadata: getConstEntry: cannot find metadata tag 917514
10-20 11:19:44.848  9545  9575 E ACameraMetadata: getConstEntry: cannot find metadata tag 917511
10-20 11:19:44.848  9545  9575 E ACameraMetadata: getConstEntry: cannot find metadata tag 917512
10-20 11:19:44.849  9545  9573 I MotionCam: Exposure state: SEARCHING
10-20 11:19:44.850  9545  9573 D libMotionCam: AUTO_FOCUS_WAIT completed
10-20 11:19:44.850  9545  9573 D libMotionCam: setState(AUTO_FOCUS_WAIT -> TRIGGER_AUTO_FOCUS)
10-20 11:19:44.850  9545  9573 D libMotionCam: nextState() [mCaptureSessionState == CameraCaptureSessionState::READY]
10-20 11:19:44.850  9545  9573 D libMotionCam: triggerAutoFocus()
10-20 11:19:44.850  9545  9545 I MotionCam: Focus state: ACTIVE_SCAN
10-20 11:19:44.851  9545  9573 D libMotionCam: Camera session has changed state (1)
10-20 11:19:44.851  9545  9573 I MotionCam: Camera state changed ACTIVE
10-20 11:19:44.852  9545  9677 W Adreno-GSL: <os_lib_map:1435>:   os_lib_map error: dlopen failed: library "libllvm-qcom.so" not found, on 'libllvm-qcom.so'
10-20 11:19:44.853  9545  9677 W Adreno-CB: <cl_osal_open_library:326>: Unable to open library: error: (null), on 'libllvm-qcom.so'
10-20 11:19:44.853  9545  9677 W Adreno-GSL: <os_lib_map:1435>:   os_lib_map error: dlopen failed: library "libllvm-qcom.so" not found, on 'libllvm-qcom.so'
10-20 11:19:44.853  9545  9677 W Adreno-CB: <cl_osal_open_library:326>: Unable to open library: error: (null), on 'libllvm-qcom.so'
10-20 11:19:44.854  9545  9677 I halide  : Error: CL: clCreateContext failed: CL_OUT_OF_HOST_MEMORY:-6
--------- beginning of crash
10-20 11:19:44.854  9545  9677 F libc    : Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 9677 (com.motioncam), pid 9545 (com.motioncam)
10-20 11:19:44.857  9545  9642 W libMotionCam: Out of buffers
10-20 11:19:44.896   733  9657 I mm-camera: <IMGLIB>< INFO> 3997: module_faceproc_client_print_config: FD CONFIG : engine=FACEPROC_ENGINE_SW, mode=1, enable_bsgc=0, chromatix id=20, chromatix used = fd_chromatix_detect_qc_moto_channel.h sw_detect=0, enable_upfront=1, sw_detect_enable_upfront=0
10-20 11:19:44.896   733  9657 I mm-camera: <IMGLIB>< INFO> 4012: module_faceproc_client_print_config: FD CONFIG : min_face_size=45, min_face_size_ratio=0.100000, speed=0, max_fps=15 (0 means max possible), thresholds= 770, 0 0 0, 0 0 0, search_cycles=10 10
10-20 11:19:44.896   733  9657 I mm-camera: <IMGLIB>< INFO> 4017: module_faceproc_client_print_config: FD CONFIG : Frame width=640, height=480, stride=640, scanline=480
10-20 11:19:44.897  9681  9681 E DEBUG   : failed to readlink /proc/9677/fd/84: No such file or directory
10-20 11:19:44.897  9681  9681 E DEBUG   : failed to readlink /proc/9677/fd/87: No such file or directory
10-20 11:19:44.898  9681  9681 E DEBUG   : failed to readlink /proc/9677/fd/90: No such file or directory
10-20 11:19:44.899   733   733 W CAM_MctBus_1: type=1400 audit(0.0:1225): avc: denied { read } for uid=1047 name="u:object_r:radio_prop:s0" dev="tmpfs" ino=16388 scontext=u:r:hal_camera_default:s0 tcontext=u:object_r:radio_prop:s0 tclass=file permissive=0
10-20 11:19:44.902   733  9581 E libc    : Access denied finding property "gsm.operator.iso-country"
10-20 11:19:44.913  9478  9478 V ClearcutMetricXmitter: Transmission is done.
10-20 11:19:44.918  9681  9681 I crash_dump64: obtaining output fd from tombstoned, type: kDebuggerdTombstone
[... cut ...]
10-20 11:19:44.933  9681  9681 F DEBUG   : backtrace:
10-20 11:19:44.933  9681  9681 F DEBUG   :       #00 pc 0000000000089acc  /apex/com.android.runtime/lib64/bionic/libc.so (abort+164) (BuildId: 48a42e30f1ee71f42e5c1d9374f3ffc1)
10-20 11:19:44.933  9681  9681 F DEBUG   :       #01 pc 0000000000561be4  /data/app/~~mUaZOk0NE4Rr18Tohg5akQ==/com.motioncam-R78ZNvqQKhtDfAFp3aHE_Q==/base.apk!libnative-camera-opencl.so (offset 0x1241000) (halide_default_error+116) (BuildId: 086703d7560ecd0f158cc69d75b6c33967c539de)
...
```

closed 2022-09-09 by genpfault