mirsadm/motioncam #23: Is it possible to use Motion Cam for RAW video?

akirayui, 2021-08-22:
Hello, 
Thank you for the project.

I was developing a similar app for personal use.
What I want to do is recording a 10-15 second DNG sequence in RAW10 format at 24FPS with 1/48 shutter speed. For this, I wanted to create a buffer and empty a certain part of it during shooting and reuse it. But I couldn't do it because it was too complicated for me.
Looks like you've already done the important part. Do you have any plans to develop your app so that it can be used for video as I mentioned?

mirsadm, 2021-08-23:
What would you want to do with the 10-15 seconds of RAW10 data? Keep in mind on a 12 MP sensor 15 seconds worth of RAW10 data is over 5GB.

locuturus, 2021-10-10:
SAF storage access framework could be used to set the save path to an SDcard or even USB storage. Open Camera has this as an option but defaults to saving under DCIM for simplicity. Assuming the storage device is fast enough to not choke this could improve the usefulness of taking even short clips since one might take a few of them at a go and rapidly fill internal storage.

mirsadm, 2021-10-19:
I've managed to get it to work. On my device (OnePlus7 Pro) I can record about 30 seconds before running out of memory. The storage is not quite fast enough to be able to keep up in real time. Still 30 seconds of footage is over 10 GB. I need to write a small tool to convert the output into DNG then I'll upload a test version.

locuturus, 2021-10-21:
Pixel 4a on Android 12. Turned off dual exposure, recorded ~9 sec (very laggy preview but no crash) and I see a zip with 177 frames inside. So far so good! No idea yet how it looks I'll have to get that all converted first.

paolodepetrillo, 2021-10-22:
I got the raw2dng utility to build and run on Arch Linux using this [CMakeLists.txt](https://gist.github.com/paolodepetrillo/a3707e83f0dcdba4e34f6804112aaf78) file. The file goes in motioncam/raw2dng. I don't really know much about build systems and got this working by trial and error, so I'm sure there are lots of things done incorrectly.

`./setupenv.sh` must have been run successfully, and OpenCV must be installed, before building this.

On [this line](https://github.com/mirsadm/motioncam/blob/adcbdfe52f0ed825d308a02189befcb2ad755196/libMotionCam/libMotionCam/include/motioncam/CameraProfile.h#L30) in CameraProfile.h I had to replace `__unused` with `__attribute__((__unused__))` to get it to compile - apparently it's defined in a OSX header but not on Linux.

I am also testing with a Pixel 4a (not the newer 5G version). Only recorded a couple seconds to try it so far.

akirayui, 2021-11-02:
The application you have published on Play Store runs without any problems, but this one crashes because Documents folder is not available by default on some Android devices like Xiaomi Redmi Note. If the user creates the directory manually,  "IOException" will thrown due to your application does not have permission to write to this directory. So I recommend using the DCIM folder reserved for media.

mirsadm, 2021-11-03:
Yep, writing to the Document folder was always a hack just to test this feature. What I intend on doing is to save to the apps private storage area first then save the DNGs out to the DCIM folder afterwards.

nongrata23, 2021-11-13:
hi, i also tried, i think is crushing because lack of permissions maybe?>

owlgreens, 2021-11-16:
> I've put up a test build here: https://github.com/mirsadm/motioncam/releases
> 
> Couple of things to note:
> 
> 1. The app records a ZIP file to the internal storage Documents folder.
> 2. The ZIP file needs to be converted to a set of DNG files. I've written a tool that will do that which is here:
>    https://github.com/mirsadm/motioncam/tree/main/bin/mac
> 
> I've only compiled it for an x86 Mac for now. I'll look into creating a Linux binary as well. Otherwise the source code for it is here: https://github.com/mirsadm/motioncam/tree/main/raw2dng

Can you please repost the zip to dng conversion tool? I'm getting a 404. Thank you and keep up the great work!

mirsadm, 2021-11-17:
I'm almost done with an update that will convert the files to DNG on the device itself after they are captured. The utility needs to be compiled at the moment. I will let you guys know when it's done.

mirsadm, 2021-11-21:
@owlgreens I've uploaded a new release here
https://github.com/mirsadm/motioncam/releases/download/6.0.3-alpha0/MotionCam.6.0.3-alpha0.apk

It will convert the video to DNG on the device after capture

@akirayui I'm hoping this will also fix your problem with the Documents folder

nongrata23, 2021-11-21:
Is possible to capture the frames into folders instead of zips?, also is possible to convert outside of the phone? because it takes a lot of time to convert even on a fast soc... but seems to be working!!

eszdman, 2021-11-22:
>That's cropping from 4000x3000 (12MP) to 4000x2100 (8.4MP).
Add scaling also, to get 3840x2160
And software binning options for 2000x1050 as example
[image]
[image]

akirayui, 2021-11-22:
The app doesn't crash anymore but I couldn't find where the video was saved.

SkyDiverCool, 2021-11-22:
[image]
Found that the last handful of frames in a zip file are at no crop

mirsadm, 2021-11-23:
New release here: [<...>](https://github.com/mirsadm/motioncam/releases/download/6.0.4-alpha0/MotionCam.6.0.4-alpha0.apk)

- That fixes the cropping issue for the last few frames
- I've added some more frame rate options (30, 24, 12, 5, 1)
- Focus is locked until clicking again on the focus circle
- Files are saved as DNGs in one directory, not in ZIP files anymore.

@eszdman The reason I've only added the cropping option is because it is very fast to do in real time. Scaling and binning would increase the chances of dropping frames. It is something you could do in post pretty easily though. Binning is an interesting idea but can also be achieved in post. If the purpose of binning is to reduce noise I can add an option to combine multiple frames.

eszdman, 2021-11-23:
@mirsadm Scaling is fast, it can decrease number of subpixels but without cropping, so you should use it to save field of view

mirsadm, 2021-11-23:
Relative to other image operations it is fast but every millisecond matters in this instance. Also skipping pixels (nearest neighbour scaling) will not result in a nice looking image. I'm wondering why you would like that though?

nongrata23, 2021-11-23:
i have no words to express what i feel hhhhhhhhhh the world will burn in the following months, get ready hhhhhhh

owlgreens, 2021-11-23:
> New release here: https://github.com/mirsadm/motioncam/releases/download/6.0.4-alpha0/MotionCam.6.0.4-alpha0.apk
> 
Gamechanging! Thank you for your incredible work!

SpinozaWeb, 2021-11-23:
I am using LG v40 on android 10. I made a RAW video. but I don't know where the files are written. I don't have a documents folder on my phone. I even created a documents folder but there are no files there. Where does the program store files?

eszdman, 2021-11-23:
> I'm wondering why you would like that though?

Because there are devices where the memory speed does not allow you to save 16 by 9, even taking into account the cropping of the edges due to the too high resolution of the source, scaling could help for this.

nozorrog, 2021-11-24:
The raw video is stored in /data/data/ root folder. It's very inusefull and need to be resolved.

Mrdoumhann, 2021-11-24:
> The raw video is stored in /data/data/ root folder. It's very unusefull and need to be resolved.

I think this folder is not accessible from non-rooted devices. Also there is no 
any other folder related to motioncam. So I could not find recorded DNGs.

nozorrog, 2021-11-24:
> > The raw video is stored in /data/data/ root folder. It's very unusefull and need to be resolved.
> 
> I think this folder is not accessible from non-rooted devices. Also there is no any other folder related to motioncam. So I could not find recorded DNGs.

You are right. And I insiste to store writed files in DCIM folder to be accessed by user without problems.

locuturus, 2021-11-24:
I suggest picking a user defined folder with SAF. Could be DCIM, Documents,
an SDcard, USB drive, whatever. My 2 cents,

87alex87, 2021-11-24:
External USB SSD (like Samsung T5) support would be great! :)

xtrul, 2021-11-24:
Again i have no words to express myself :) i still cannot recover of excitement! still many things can be improved but the base is here now... i left you several messages let me know when u want can till then i enjoy this as it is

Regards
Sebastian

[Here is a sample](https://youtu.be/Ww4r0KUUYE4)

mirsadm, 2021-11-25:
Update here:
https://github.com/mirsadm/motioncam/releases/download/6.0.5-alpha0/MotionCam.6.0.5-alpha0.apk

Videos are saved to Downloads folder now instead.

Mrdoumhann, 2021-11-25:
> Update here: https://github.com/mirsadm/motioncam/releases/download/6.0.5-alpha0/MotionCam.6.0.5-alpha0.apk
> 
> Videos are saved to Downloads folder now instead.

Thanks for the update. But I can't find Downloads folder. Whats wrong with DCIM?

mirsadm, 2021-11-25:
Storing them in the DCIM folder will result in every video frame showing up in Google Photos/image gallery which IMO is not good. Have you tried using a file manager? I'd be shocked if your phone doesn't have a Download folder

xtrul, 2021-11-25:
is working fine, on downloads folder for s9+, lg g7, s20 fe^^

Mrdoumhann, 2021-11-25:
[image]

Yes Download folder is there but I think the app is looking for the "Download**s**"folder.

mirsadm, 2021-11-26:
The app doesn't look for a Downloads or Download folder. It asks the OS to save it to wherever the Download folder is.

New release: here https://github.com/mirsadm/motioncam/releases/download/6.0.6-alpha0/MotionCam.6.0.6-alpha0.apk

- RAW videos don't get processed after capture now. They can manually be processed at a later time.
- If dual exposure is enabled in settings, RAW video capture mode provides a preview of the RAW image with a clipping indicator.
-  And finally the app shows the resolution and provides more cropping options.

xtrul, 2021-11-26:
Thank you for this solid update! works great all features!

Mrdoumhann, 2021-11-27:
> The app doesn't look for a Downloads or Download folder. It asks the OS to save it to wherever the Download folder is.

There are enough testers already, so just ignore this issue for now. Development is going very well, keep it up.

mirsadm, 2021-11-27:
One final update for a while here: https://github.com/mirsadm/motioncam/releases/download/6.0.7-alpha0/MotionCam.6.0.7-alpha0.apk

@Mrdoumhann I've changed it so it lets you select where to save the RAW videos rather than going to Documents/Download. Hopefully that fixes your issue.

Mrdoumhann, 2021-11-27:
> One final update for a while here: https://github.com/mirsadm/motioncam/releases/download/6.0.7-alpha0/MotionCam.6.0.7-alpha0.apk
> 
> @Mrdoumhann I've changed it so it lets you select where to save the RAW videos rather than going to Documents/Download. Hopefully that fixes your issue.

 Yes, problem solved. Finally I am able to see the holy dngs. Thank you for your effort.

sx3z56, 2021-11-28:
> One final update for a while here: https://github.com/mirsadm/motioncam/releases/download/6.0.7-alpha0/MotionCam.6.0.7-alpha0.apk
> 
> @Mrdoumhann I've changed it so it lets you select where to save the RAW videos rather than going to Documents/Download. Hopefully that fixes your issue.

Hey! thank you very much for your work! it gives new life to mobile video shooting!I have a question. at the moment, your crop is implemented only along the x-axis. do you plan to add a crop on the y-axis? the thing is, my phone's main sensor has an 27 megapixels. and even with the maximum memory allocation, the buffer is clogged in seconds. a crop on the other axis would help a lot.

mirsadm, 2021-11-29:
Yep I've added both horizontal/vertical cropping and fixed an issue with purple DNGs. Just need a bit more time to sort it out.

experimentator1709, 2021-11-30:
> Yep I've added both horizontal/vertical cropping and fixed an issue with purple DNGs. Just need a bit more time to sort it out.

Hey! Please add 48, 50 and 60 frames per second. Another question is whether it is possible to record sound, at least in WAV. Thanks!

xtrul, 2021-11-30:
> > Yep I've added both horizontal/vertical cropping and fixed an issue with purple DNGs. Just need a bit more time to sort it out.
> 
> Hey! Please add 48, 50 and 60 frames per second. Another question is whether it is possible to record sound, at least in WAV? Thanks!

48 50 60 wont work! unless the oem set it free

experimentator1709, 2021-11-30:
> 48 50 60 wont work! unless the oem set it free

Thanks for the information. But suddenly the author succeeds :)

xtrul, 2021-11-30:
Thank you! @mirsadm(for your brains) @akirayui (for yourvision) let the revolution begin!

https://youtu.be/slUvVN3VqJY

mirsadm, 2021-11-30:
Update here https://github.com/mirsadm/motioncam/releases/download/6.0.8-alpha0/MotionCam.6.0.8-alpha0.apk

- Horizontal/vertical cropped
- Purple DNGs fix
- UI improvements 
- AE/AWB lock and OIS toggle

mirsadm, 2021-12-03:
Update here https://github.com/mirsadm/motioncam/releases/download/7.0.1/MotionCam.7.0.1.apk
AF locking and some bug fixes

xtrul, 2021-12-03:
> > Update here https://github.com/mirsadm/motioncam/releases/download/7.0.1/MotionCam.7.0.1.apk AF locking and some bug fixes
> 
> the buttons are so small

Lets finish functionality first :) UI after

yousaf465, 2021-12-03:
you can edit the resulting dng directly in Da Vinci resolve by unchecking "Show Individual Frames" https://beginnersapproach.com/davinci-resolve-edit-images-stills/

mirsadm, 2021-12-03:
https://github.com/mirsadm/motioncam/releases/download/7.0.2/MotionCam.7.0.2.apk

Just a small update to make the buttons a little bit easier to click

experimentator1709, 2021-12-04:
> https://github.com/mirsadm/motioncam/releases/download/7.0.2/MotionCam.7.0.2.apk
> 
> Just a small update to make the buttons a little bit easier to click

HI! Which API did all cameras become available?

mirsadm, 2021-12-04:
They're not really available, the app guesses and tries to open more cameras than reported by the system. This usually let's it use more of the cameras than reported by the system

experimentator1709, 2021-12-04:
What is the quality of the JPG preview inside the DNG? Can we save the size of the DNG file? For example 50% JPG quality for preview?

Automatisator100, 2021-12-04:
Hello Mirsad,

this is a super project (many are waiting for it!) and it is discussed in cinematography forums. 

The Huawei smartphones have a RYYB sensor, could you implement a conversion to RGGB before writing to DNG? That would be great. :) Because then the DNG from huawei can be edited directly in Davinci.

Best regards

julienacquaviva, 2021-12-05:
Hello Mirsad,

Thanks a lot for you work, RAW video works very very well.
Although, I have hot pixels on my DNGs. Is there something fixable? Thanks in advance

[image]

mirsadm, 2021-12-05:
@experimentator1709 do you mean reducing the size of the DNG? I can enable compression when creating the DNGs. It will make the process take longer though. Eventually I want to be able to do the conversion on a PC to reduce the time it takes.

@Automatisator100 It is certainly possible though I don't have access to a Huawei phone so it would be difficult for me to test.

@julienacquaviva I think those dots might from the auto focus system. Try the release below and set the frame merging to 4 or higher. If that doesn't fix it then I'll add the hot pixel removal that I use for photos.

Update here:
https://github.com/mirsadm/motioncam/releases/download/7.0.3/MotionCam.7.0.3.apk

- Option to merge frames to reduce noise when converting RAW video
- UI fixes

experimentator1709, 2021-12-05:
@mirsadm Thank you! I'm waiting for the conversion to PC

experimentator1709, 2021-12-05:
@mirsadm Can remove vignette be added to RAW metadata?

julienacquaviva, 2021-12-05:
@mirsadm thanks for that quick answer, frame merging doesn't fix it though.
First I thought it was focus pixels then I tried Rawtherapee and applied the "hot pixels" fix and it worked.
Btw those dots come with the RAW photo mode as well (I mean Motion Cam's photo mode) but not with RAW photo from the stock camera.

Automatisator100, 2021-12-05:
> @Automatisator100 It is certainly possible though I don't have access to a Huawei phone so it would be difficult for me to test.

I would offer myself as a tester. :)

But I think it is a bug. When I activate "Enable dual exposure controls", I have a picture like this:

https://drive.google.com/file/d/1S4wwXY4UbseGyk9TWY7mRO3oa9reH1cn/view?usp=sharing

if I deactivate "dual exposure controls" everything fits:

https://drive.google.com/file/d/1S0WdjdZDr1_F2_mZbombb3roFSzm4ZvD/view?usp=sharing

And when I use RAW VIDEO, it looks exactly the same as in PHOTO mode with "Enable dual exposure controls" enabled.

mirsadm, 2021-12-06:
@Automatisator100 It is a bug in the sense that it doesn't support that bayer layout. It only supports RGGB/BGGR/GRGB/GBGR. When you enable the dual exposure controls it doesn't use the Android native preview, it generates one from the RAW images.

Update here with general UI fixes:
https://github.com/mirsadm/motioncam/releases/download/7.0.4/MotionCam.7.0.4.apk

[33 more replies]