mirsadm/motioncam #26: v5 worse dynamic range

Sergio677, 2021-10-03:
As I understand, v4 uses underexposure and increases shadows. HDR effect in result.

Beginning with v5 there is no hdr. Also, lacks contrast and sharpening (yes, more noise, but overall looks better, crisper).

[image]
[image]

mirsadm, 2021-10-04:
In general I've tried to tone down the amount of processing to make the images look a bit more natural. I will add a setting for sharpness so that it can be adjusted. I've pushed out a new release that increases the aggressiveness of HDR. It's on the releases page now. Let me know if that helps, in general the HDR should be much better than in v4.

Sergio677, 2021-10-05:
5.0.3 / 5.0.1 / 4.0.0
[image]
[image]
[image]
It's strange but now there is no strong difference between 4 & 5, only cosmetic. Maybe because of iso/shutter are similar in all 3 images. Earlier v4 always underexposed in compare to 5

closed 2021-11-03 by mirsadm