mirsadm/motioncam #29: Request

KirChi123, 2021-11-10:
Hello! Please if you can make MotionCam available for ios because there is no such app for ios and competitive to your algorithm!

MDNSSKNGHT, 2021-11-10:
I'm not that in ios development but if I'm sure, the lang for coding isn't compatible with JNI (mostly the algorithms are in the shared libraries) So he would need to code it from zero for ios

He can use Java, but as I'm seeing, it has such poor performance and requires significant memory space

mirsadm, 2021-11-10:
All of the algorithms are written in C++ so it is possible to make it work with iOS. It is something I am considering, it'll just take some time.

KirChi123, 2021-11-10:
Please make it because my 6s needs conputational photography even if it is a more simple computational photograohy hdr app. You can even use this open source code: https://github.com/martin-marek/hdr-plus-swift

MDNSSKNGHT, 2021-11-10:
@mirsadm oh yeah you can basically code using C++ in ios, but won't it take some quite time to get done up to upstream from android one?