mirsadm/motioncam #30: Just crash on startup

dimka11, 2021-11-28:
Meizu 16X Android 8.1

logcat:

```
11-28 04:15:45.513 29273 29273 E AndroidRuntime: FATAL EXCEPTION: main
11-28 04:15:45.513 29273 29273 E AndroidRuntime: Process: com.motioncam, PID: 29273
11-28 04:15:45.513 29273 29273 E AndroidRuntime: java.lang.RuntimeException: Unable to resume activity {com.motioncam/com.motioncam.CameraActivity}: java.lang.NullPointerException: Attempt to read from field 'java.lang.String com.motioncam.camera.NativeCameraInfo.cameraId' on a null object reference
11-28 04:15:45.513 29273 29273 E AndroidRuntime: 	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3780)
11-28 04:15:45.513 29273 29273 E AndroidRuntime: 	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:3820)
11-28 04:15:45.513 29273 29273 E AndroidRuntime: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1778)
11-28 04:15:45.513 29273 29273 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:106)
11-28 04:15:45.513 29273 29273 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:164)
11-28 04:15:45.513 29273 29273 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:6758)
11-28 04:15:45.513 29273 29273 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
11-28 04:15:45.513 29273 29273 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
11-28 04:15:45.513 29273 29273 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:886)
11-28 04:15:45.513 29273 29273 E AndroidRuntime: Caused by: java.lang.NullPointerException: Attempt to read from field 'java.lang.String com.motioncam.camera.NativeCameraInfo.cameraId' on a null object reference
11-28 04:15:45.513 29273 29273 E AndroidRuntime: 	at com.motioncam.camera.NativeCameraSessionBridge.getMetadata(NativeCameraSessionBridge.java:329)
11-28 04:15:45.513 29273 29273 E AndroidRuntime: 	at com.motioncam.CameraActivity.initCamera(CameraActivity.java:1281)
11-28 04:15:45.513 29273 29273 E AndroidRuntime: 	at com.motioncam.CameraActivity.onResume(CameraActivity.java:559)
11-28 04:15:45.513 29273 29273 E AndroidRuntime: 	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1357)
11-28 04:15:45.513 29273 29273 E AndroidRuntime: 	at android.app.Activity.performResume(Activity.java:7160)
11-28 04:15:45.513 29273 29273 E AndroidRuntime: 	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3755)
11-28 04:15:45.513 29273 29273 E AndroidRuntime: 	... 8 more
11-28 04:15:47.542 29352 29352 E AndroidRuntime: FATAL EXCEPTION: main
11-28 04:15:47.542 29352 29352 E AndroidRuntime: Process: com.motioncam, PID: 29352
11-28 04:15:47.542 29352 29352 E AndroidRuntime: java.lang.RuntimeException: Unable to resume activity {com.motioncam/com.motioncam.CameraActivity}: java.lang.NullPointerException: Attempt to read from field 'java.lang.String com.motioncam.camera.NativeCameraInfo.cameraId' on a null object reference
11-28 04:15:47.542 29352 29352 E AndroidRuntime: 	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3780)
11-28 04:15:47.542 29352 29352 E AndroidRuntime: 	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:3820)
11-28 04:15:47.542 29352 29352 E AndroidRuntime: 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3044)
11-28 04:15:47.542 29352 29352 E AndroidRuntime: 	at android.app.ActivityThread.-wrap11(Unknown Source:0)
11-28 04:15:47.542 29352 29352 E AndroidRuntime: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1729)
11-28 04:15:47.542 29352 29352 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:106)
11-28 04:15:47.542 29352 29352 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:164)
11-28 04:15:47.542 29352 29352 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:6758)
11-28 04:15:47.542 29352 29352 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
11-28 04:15:47.542 29352 29352 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
11-28 04:15:47.542 29352 29352 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:886)
11-28 04:15:47.542 29352 29352 E AndroidRuntime: Caused by: java.lang.NullPointerException: Attempt to read from field 'java.lang.String com.motioncam.camera.NativeCameraInfo.cameraId' on a null object reference
11-28 04:15:47.542 29352 29352 E AndroidRuntime: 	at com.motioncam.camera.NativeCameraSessionBridge.getMetadata(NativeCameraSessionBridge.java:329)
11-28 04:15:47.542 29352 29352 E AndroidRuntime: 	at com.motioncam.CameraActivity.initCamera(CameraActivity.java:1281)
11-28 04:15:47.542 29352 29352 E AndroidRuntime: 	at com.motioncam.CameraActivity.onResume(CameraActivity.java:559)
11-28 04:15:47.542 29352 29352 E AndroidRuntime: 	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1357)
11-28 04:15:47.542 29352 29352 E AndroidRuntime: 	at android.app.Activity.performResume(Activity.java:7160)
11-28 04:15:47.542 29352 29352 E AndroidRuntime: 	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3755)
11-28 04:15:47.542 29352 29352 E AndroidRuntime: 	... 10 more
11-28 04:18:11.332 31764 31764 E AndroidRuntime: FATAL EXCEPTION: main
11-28 04:18:11.332 31764 31764 E AndroidRuntime: Process: com.motioncam, PID: 31764
11-28 04:18:11.332 31764 31764 E AndroidRuntime: java.lang.RuntimeException: Unable to resume activity {com.motioncam/com.motioncam.CameraActivity}: java.lang.NullPointerException: Attempt to read from field 'java.lang.String com.motioncam.camera.NativeCameraInfo.cameraId' on a null object reference
11-28 04:18:11.332 31764 31764 E AndroidRuntime: 	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3780)
11-28 04:18:11.332 31764 31764 E AndroidRuntime: 	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:3820)
11-28 04:18:11.332 31764 31764 E AndroidRuntime: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1778)
11-28 04:18:11.332 31764 31764 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:106)
11-28 04:18:11.332 31764 31764 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:164)
11-28 04:18:11.332 31764 31764 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:6758)
11-28 04:18:11.332 31764 31764 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
11-28 04:18:11.332 31764 31764 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
11-28 04:18:11.332 31764 31764 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:886)
11-28 04:18:11.332 31764 31764 E AndroidRuntime: Caused by: java.lang.NullPointerException: Attempt to read from field 'java.lang.String com.motioncam.camera.NativeCameraInfo.cameraId' on a null object reference
11-28 04:18:11.332 31764 31764 E AndroidRuntime: 	at com.motioncam.camera.NativeCameraSessionBridge.getMetadata(NativeCameraSessionBridge.java:329)
11-28 04:18:11.332 31764 31764 E AndroidRuntime: 	at com.motioncam.CameraActivity.initCamera(CameraActivity.java:1484)
11-28 04:18:11.332 31764 31764 E AndroidRuntime: 	at com.motioncam.CameraActivity.onResume(CameraActivity.java:599)
11-28 04:18:11.332 31764 31764 E AndroidRuntime: 	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1357)
11-28 04:18:11.332 31764 31764 E AndroidRuntime: 	at android.app.Activity.performResume(Activity.java:7160)
11-28 04:18:11.332 31764 31764 E AndroidRuntime: 	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3755)
11-28 04:18:11.332 31764 31764 E AndroidRuntime: 	... 8 more
11-28 04:18:19.412 31970 31970 E AndroidRuntime: FATAL EXCEPTION: main
11-28 04:18:19.412 31970 31970 E AndroidRuntime: Process: com.motioncam, PID: 31970
11-28 04:18:19.412 31970 31970 E AndroidRuntime: java.lang.RuntimeException: Unable to resume activity {com.motioncam/com.motioncam.CameraActivity}: java.lang.NullPointerException: Attempt to read from field 'java.lang.String com.motioncam.camera.NativeCameraInfo.cameraId' on a null object reference
[... 267 lines cut ...]
11-28 04:32:06.998 11774 11774 E AndroidRuntime: 	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3755)
11-28 04:32:06.998 11774 11774 E AndroidRuntime: 	... 10 more
[... cut ...]
11-28 04:32:28.754 12212 12212 E AndroidRuntime: 	at com.motioncam.CameraActivity.onResume(CameraActivity.java:599)
11-28 04:32:28.754 12212 12212 E AndroidRuntime: 	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1357)
11-28 04:32:28.754 12212 12212 E AndroidRuntime: 	at android.app.Activity.performResume(Activity.java:7160)
11-28 04:32:28.754 12212 12212 E AndroidRuntime: 	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3755)
11-28 04:32:28.754 12212 12212 E AndroidRuntime: 	... 10 more
```

dimka11, 2021-11-28:
```
8 15:23:33.420  1385  8964 I ActivityManager: START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10200000 cmp=com.motioncam/.CameraActivity bnds=[0,348][205,624]} from uid 10116
11-28 15:23:33.423   658   724 E ANDR-PERF-MPCTL: Invalid profile no. 0, total profiles 0 only
11-28 15:23:33.425   519   519 W /system/bin/hwservicemanager: getTransport: Cannot find entry vendor.qti.hardware.iop@2.0::IIop/default in either framework or device manifest.
11-28 15:23:33.429   658   724 E ANDR-PERF-OPTSHANDLER: Perflock resource /sys/class/devfreq/soc:qcom,llccbw/min_freq not supported
11-28 15:23:33.429   658   724 E ANDR-PERF-RESOURCEQS: Failed to apply optimization [12, 0]
11-28 15:23:33.435  1385  1641 I chatty  : uid=1000(system) android.anim expire 6 lines
11-28 15:23:33.460   677   677 I BufferQueue: [unnamed-677-1753](this:0x7cd4388800,id:1753,api:0,p:-1,c:-1) BufferQueue core=(677:/system/bin/surfaceflinger)
11-28 15:23:33.469  2556  3303 W Ims: b: : Skipped handle as screen on!
11-28 15:23:33.487  1385  1724 I chatty  : uid=1000(system) scene_boost expire 2 lines
11-28 15:23:33.494  2783  2783 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.504  2783  2783 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.545 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.553 26673 26673 I chatty  : uid=10364(com.motioncam) identical 321 lines
11-28 15:23:33.582 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.582  1797  1797 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.583  1797  1797 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.589  2303  2303 W recents.Component: create a new LoadPlan to load thumbnail -- background
11-28 15:23:33.591 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.604 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.605  1385  1664 I chatty  : uid=1000(system) RenderThread expire 1 line
11-28 15:23:33.605 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.620 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.620  2303  2303 I recents.TaskLoadPlan: 		 load thumbnail from system
11-28 15:23:33.620 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.636 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.637  2303  2303 I recents.TaskLoadPlan: 		 load thumbnail from system
11-28 15:23:33.637 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.653 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.653  2303  2303 I recents.TaskLoadPlan: 		 load thumbnail from system
11-28 15:23:33.654 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.670 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.674  2303  2303 I recents.TaskLoadPlan: 		 load thumbnail from system
11-28 15:23:33.678 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.691 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.693  2303  2303 I recents.TaskLoadPlan: 		 load thumbnail from system
11-28 15:23:33.693  2303  2303 W recents.Performance: preload while task Change spend : 104
11-28 15:23:33.694 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.700 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.700 26673 26678 I zygote64: Do partial code cache collection, code=29KB, data=26KB
11-28 15:23:33.701 26673 26678 I zygote64: After code cache collection, code=29KB, data=26KB
11-28 15:23:33.701 26673 26678 I zygote64: Increasing code cache capacity to 128KB
11-28 15:23:33.701 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.702  1385  1424 E LmsInjector: removeUpdatesForPowerSaveInner: failed java.lang.NullPointerException: Attempt to invoke interface method 'java.lang.Object[] java.util.Collection.toArray()' on a null object reference
11-28 15:23:33.734 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.758  1385  2088 I chatty  : uid=1000(system) Binder:1385_9 expire 1 line
11-28 15:23:33.760   646   646 I chatty  : uid=1000(system) sensors@1.0-ser expire 12 lines
11-28 15:23:33.764  1385  8964 I chatty  : uid=1000(system) Binder:1385_1E expire 1 line
11-28 15:23:33.768 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.771 26673 26673 W linker  : "/data/app/com.motioncam-OZIjJmLHXat2Rog7bb6i1w==/base.apk!/lib/arm64-v8a/libc++_shared.so" unused DT entry: type 0x70000001 arg 0x0
11-28 15:23:33.785 26673 26673 I libMotionCam: Device has 2 exposed camera
11-28 15:23:33.785   940   940 E CameraProviderManager: pCameraIdStr(0) packageName()
11-28 15:23:33.787   940   940 E CameraProviderManager: pCameraIdStr(1) packageName()
11-28 15:23:33.788   940   940 E CameraProviderManager: pCameraIdStr(2) packageName()
11-28 15:23:33.790   940   940 E CameraProviderManager: pCameraIdStr(3) packageName()
11-28 15:23:33.791   940   940 E CameraProviderManager: pCameraIdStr(4) packageName()
11-28 15:23:33.792 26673 26673 I libMotionCam: Found 0 supported cameras
11-28 15:23:33.794 26673 26673 I FlymeFontsHelper: UserId == 0
11-28 15:23:33.827 26673 26673 E AndroidRuntime: FATAL EXCEPTION: main
11-28 15:23:33.827 26673 26673 E AndroidRuntime: Process: com.motioncam, PID: 26673
11-28 15:23:33.827 26673 26673 E AndroidRuntime: java.lang.RuntimeException: Unable to resume activity {com.motioncam/com.motioncam.CameraActivity}: java.lang.NullPointerException: Attempt to read from field 'java.lang.String com.motioncam.camera.NativeCameraInfo.cameraId' on a null object reference
[... cut ...]
11-28 15:23:33.827 26673 26673 E AndroidRuntime: 	at com.motioncam.CameraActivity.onResume(CameraActivity.java:599)
11-28 15:23:33.827 26673 26673 E AndroidRuntime: 	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1357)
11-28 15:23:33.827 26673 26673 E AndroidRuntime: 	at android.app.Activity.performResume(Activity.java:7160)
11-28 15:23:33.827 26673 26673 E AndroidRuntime: 	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3755)
11-28 15:23:33.827 26673 26673 E AndroidRuntime: 	... 10 more
```

mirsadm, 2021-12-08:
This crash is fixed but unfortunately it was happening because the app couldn't find any supported cameras.

Update here:
https://github.com/mirsadm/motioncam/releases/download/7.0.6/MotionCam.7.0.6.apk

closed 2021-12-08 by mirsadm

dimka11, 2021-12-08:
@mirsadm but why? GCam, FreeDCam, Photon and many other camera app's can capture raw on my device.