mirsadm/motioncam #31: Add instruction for how build app under Windows

dimka11, 2021-12-01:
I need build this app under Windows

joemarshall, 2021-12-17:
You can build it in Windows subsystem for Linux. Just install Ubuntu in wsl and then follow the Ubuntu instructions.

dimka11, 2021-12-17:
@joemarshall but, how install Android Studio in WSL?

joemarshall, 2021-12-17:
You could probably install android studio in WSL and run it nowadays (although you lose emulator support that way), but you can build an android studio project by just running the gradlew script in the top of the project (i.e. with only the android sdks/ndks and tools installed, not full android studio)

mirsadm, 2021-12-23:
I don't have a Windows machine to be able to set this up. It should be possible to get it to work with some effort. Unfortunately I don't have the time to do this now

closed 2021-12-23 by mirsadm