mirsadm/motioncam #35: [Feature request] - simple zebra for overexposure

Thundermaxx, 2021-12-04:
Version 7.02 absolutely nails it on my s20u. This is officially the best photo app for the exynos version! Focus is now working perfectly and doesn't hang anymore and the af lock is a fine addition. Low light performance up to 2000 iso is magic, noiseless but still with plenty of detail and no trace of ugly compression or noise reduction artefacts. HDR processing is high class. Exposure, colors, tones, sharpening, it's all so natural. Would be great to have a sharpening setting in photo mode too as I feel I could turn it very slightly down sometimes. Only with very high contrast scenes is it often overexposing. This can luckily be adjusted with the highlight slider and simple zebra could then be of great help to assess the exposure on screen but I think you could be more conservative with the highlights in auto mode. Apart from this, the only little problem left is that, no matter the settings, i still see little blocks flickering and flying around on the screen in preview mode though it doesnt't show up in the pictures. And there is color banding in very deep shadows but this happens with other apps too so I don't think it can be fixed. Anyway, fantastic work ☺

mirsadm, 2021-12-06:
Thanks for the feedback! I think in a (near) future release I'll match the burst mode settings to the normal photo ones so that they can be configured/saved. I did update the app recently and it should reduce the overexposure issue. 

Here:
https://github.com/mirsadm/motioncam/releases/download/7.0.4/MotionCam.7.0.4.apk

Thundermaxx, 2021-12-06:
Hey !

Thanks for the fast update pace and dramatic improvements ☺.
Just don’t alter the general HDR balance (color saturation, mapping, sharpness) as it is spot on.

I eventually noticed that the white circle serves for exposure metering, so now i can get the right exposure by moving this circle over the brightest part of the scene then locking AE.

Actually the problem is the S20U’s sensor and its poor dynamic range. I think the overexposed image used to recover the shadows still isn’t bright enough to avoid color banding (only happening with  darker scenes and high dynamic range).
In attachment, find an example of this color banding.

I’m gonna try 7.05 but still very happy with 7.02

I see there is currently a lot of traction for your app, so keep up the good work.

Stephan Buchin
Brussels, Belgium

https://github.com/mirsadm/motioncam/releases/download/7.0.4/MotionCam.7.0.4.apk

mirsadm, 2021-12-24:
I've added an exposure clipping indicator in the latest beta

closed 2021-12-24 by mirsadm