mirsadm/motioncam #37: [Feature request] - Using external SSD drive

87alex87, 2021-12-06:
Hi.

Is it possible to write raw files to OTG external drive instead of internal phone memory? (Rationale: saving internal flash memory from being constantly overwritten)

Thanks.

mirsadm, 2021-12-10:
Working on this one now!

mirsadm, 2021-12-11:
There's a test build of this here:
https://github.com/mirsadm/motioncam/releases/download/7.0.8-beta/MotionCam.7.0.8-beta.apk

87alex87, 2021-12-11:
Hi. When I select an external usb drive folder in the settings I get this error:

`2021-12-11 19:54:20.211 5993-5993/? E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.motioncam, PID: 5993
    java.lang.SecurityException: No permission grants found for UID 10410 and Uri 
        at android.os.Parcel.createExceptionOrNull(Parcel.java:2385)
        at android.os.Parcel.createException(Parcel.java:2369)
        at android.os.Parcel.readException(Parcel.java:2352)
        at android.os.Parcel.readException(Parcel.java:2294)
        at android.app.IUriGrantsManager$Stub$Proxy.releasePersistableUriPermission(IUriGrantsManager.java:308)
        at android.content.ContentResolver.releasePersistableUriPermission(ContentResolver.java:2898)
        at com.motioncam.ui.SettingsFragment.lambda$onAttach$10$com-motioncam-ui-SettingsFragment(SettingsFragment.java:129)
        at com.motioncam.ui.SettingsFragment$$ExternalSyntheticLambda3.onActivityResult(Unknown Source:4)
        at androidx.activity.result.ActivityResultRegistry$1.onStateChanged(ActivityResultRegistry.java:148)
        at androidx.lifecycle.LifecycleRegistry$ObserverWithState.dispatchEvent(LifecycleRegistry.java:354)
        at androidx.lifecycle.LifecycleRegistry.forwardPass(LifecycleRegistry.java:265)
        at androidx.lifecycle.LifecycleRegistry.sync(LifecycleRegistry.java:307)
        at androidx.lifecycle.LifecycleRegistry.moveToState(LifecycleRegistry.java:148)
        at androidx.lifecycle.LifecycleRegistry.handleLifecycleEvent(LifecycleRegistry.java:134)
        at androidx.fragment.app.Fragment.performStart(Fragment.java:3026)
        at androidx.fragment.app.FragmentStateManager.start(FragmentStateManager.java:589)
        at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:300)
        at androidx.fragment.app.FragmentStore.moveToExpectedState(FragmentStore.java:112)
        at androidx.fragment.app.FragmentManager.moveToState(FragmentManager.java:1647)
        at androidx.fragment.app.FragmentManager.dispatchStateChange(FragmentManager.java:3128)
        at androidx.fragment.app.FragmentManager.dispatchStart(FragmentManager.java:3079)
        at androidx.fragment.app.FragmentController.dispatchStart(FragmentController.java:262)
        at androidx.fragment.app.FragmentActivity.onStart(FragmentActivity.java:510)
        at androidx.appcompat.app.AppCompatActivity.onStart(AppCompatActivity.java:246)
        at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1435)
        at android.app.Activity.performStart(Activity.java:8231)
        at android.app.ActivityThread.handleStartActivity(ActivityThread.java:3861)
        at android.app.servertransaction.TransactionExecutor.performLifecycleSequence(TransactionExecutor.java:221)
        at android.app.servertransaction.TransactionExecutor.cycleToPath(TransactionExecutor.java:201)
        at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:173)
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:97)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2325)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:246)
        at android.app.ActivityThread.main(ActivityThread.java:8633)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:602)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)
     Caused by: android.os.RemoteException: Remote stack trace:
        at com.android.server.uri.UriGrantsManagerService.releasePersistableUriPermission(UriGrantsManagerService.java:426)
        at android.app.IUriGrantsManager$Stub.onTransact(IUriGrantsManager.java:159)
        at android.os.Binder.execTransactInternal(Binder.java:1190)
        at android.os.Binder.execTransact(Binder.java:1159)
`

mirsadm, 2021-12-11:
Yep fixed this will update soonish

mirsadm, 2021-12-11:
This should be fixed here:
https://github.com/mirsadm/motioncam/releases/download/7.0.8-beta1/MotionCam.7.0.8-beta1.apk

87alex87, 2021-12-12:
Thanks! I was able to record 3 minutes of video at 25% crop (4032 x 2268) at 24 fps without dropped frames to the external SSD drive (Samsung T5).

mirsadm, 2021-12-13:
The crashes for this should finally be fixed! @87alex87 I'm glad that worked well for you. Check out the update here:
https://github.com/mirsadm/motioncam/releases/download/7.0.8-beta2/MotionCam.7.0.8-beta2.apk

I've made some performance improvements which should reduce frame drops. You can also increase the memory for RAW video to a higher value now.

closed 2021-12-13 by mirsadm