mirsadm/motioncam #38: Make app multilanguage

bovirus, 2021-12-07:
@mirsadm

Could be possible to make your app multilanguage.

It's enough to create in

https://github.com/mirsadm/motioncam/tree/main/MotionCam-Android/MotionCam/src/main/res

a folder **values-xx** 

where **xx** it's language code (ex. **values-it** for Italian) and put in this folder a file

s**trings.xml**

that contain translation for original **values\strings.xml** file.

Thanks. KR.

mirsadm, 2021-12-08:
Thanks for submitting the translations. They've been added and should be in the next release:
https://github.com/mirsadm/motioncam/releases/download/7.0.6/MotionCam.7.0.6.apk

I've added some more strings that were hard coded.

closed 2021-12-08 by mirsadm

bovirus, 2021-12-08:
@mirsadm 

Thanks for new version.

Updated italian language

https://github.com/mirsadm/motioncam/pull/45

Some proposal for the future.

When you will make some changes on master language files take some time befroe to recompiel the application.
In this case translators like me (I hope that other guys add new languages) have the time to update language files.

Because in this acse tyou relased a new version but Italian language was not updated.

Please consider to move language files on some free project like Transiflex.

can be an enachement about translation management.