mirsadm/motioncam #42: [RAW video] Crash on tapping "Manage Videos" if a recording failed

epicfilmmaker93, 2021-12-07:
If a recording failed and crashed the app, clicking "Manage Videos" will also crash the app and make all saved videos unaccessible

newillusion, 2021-12-08:
I confirm issue, i have same situation in Samsung Note 8 (Android 9). And, crash happened, when broken video record displayed on screen. About ten videos are in my storage, and several upper ones are displayed, and, after that, program crached after moment.

And there is no tool to delete broken video records from internal application memory. Is it possible to make video records in accesible application memory, for excluse such situation? If use accesible memory, users can delete broken records manually...

mirsadm, 2021-12-08:
I've been able to reproduce this. Will put out an update today.

slonick81, 2021-12-08:
Yes, bumped on it on Poco X3, too. Same crash also happens when internal storage is full - can not delete clips to free space, only uninstall the whole app. BTW, is it possible to store intermediate clips in some more public place and have direct access to them? I mean it's internal storage anyway so no performance penalties, right?

mirsadm, 2021-12-08:
Yeah it is. I'll look at adding that to the settings. It should also be possible to use external storage (e.g. fast SSD) to remove the IO bottleneck.

mirsadm, 2021-12-08:
This should be fixed here (and Google Play release later today):
https://github.com/mirsadm/motioncam/releases/download/7.0.6/MotionCam.7.0.6.apk

closed 2021-12-08 by mirsadm