mirsadm/motioncam #44: [RAW video] Frame reordering on segment border

slonick81, 2021-12-08:
Currently big raw image sequences are split into segments/chunks of 900 frames. Problem: frames are written in wrong order to the chunks for some transitional duration (about 50 frames). Here is an actual reconstruction of such transition and some kind of scheme to illustrate:
[image]
[image]
Frames inside the chunk are sequential in historical/time sense.
Looks like there is some kind of multithreaded independent output or buffering, and some threads are still writing to previous chunk while others have already started a new one.
As a suggestion I would ask to make global frame counter through the chunks for each shot (and some unique name, like "prefix+date+start_time", not "frame-" for everything). I tend to unite all sequence files of single shot in one directory anyway for ease of editing, so as long as global frame numbering is correct I would not bother about frame order inside chunks.

mirsadm, 2021-12-08:
Thanks will look into this. Quite likely it is happening when it flushes the buffers at the end, a global frame counter is a good idea.

mirsadm, 2021-12-13:
This should be fixed. ZIP files are not longer segmented and frames are converted based on time recorded by the camera.
Update here:
https://github.com/mirsadm/motioncam/releases/download/7.0.8-beta2/MotionCam.7.0.8-beta2.apk

closed 2021-12-13 by mirsadm