mirsadm/motioncam #46: Xiaomi Mi 11

ptcptc, 2021-12-08:
Live view freeze after 2 -3 seconds on xiaomi mi 11.

Mo-body, 2021-12-08:
Same issue on Asus Zenfone 8. Android 11

mirsadm, 2021-12-08:
Seems to be something to do with Snapdragon 888 phones. I don't have one to test with though.

Does it only happen with dual exposure controls enabled in settings or in all cases?

Could you try selecting RAW16 in settings, that may help.

ptcptc, 2021-12-08:
Any setting same result. I really respect your work. You have made something really beautiful for us filmmakers. The raw video feature is amazing

mirsadm, 2021-12-09:
Thanks for the kind feedback! I'm looking into this, just need to source the right device.

ishansd94, 2021-12-11:
@mirsadm I have also opened a issue regarding the S21 Ultra(Snapdragon 888). I believe it's the same case as this. I'm more than happy to beta test the application. This is some top notch work.

mirsadm, 2021-12-11:
That would be very helpful. Do you know how to capture logs from adb?

ishansd94, 2021-12-11:
@mirsadm I'm familiar with the adb command, but I haven't used it. If you can point me on where to start I sure that I can figure out the rest.

mirsadm, 2021-12-11:
Yep here's another one I found
https://support.citrix.com/article/CTX232375

If you could send me that would be very helpful

ishansd94, 2021-12-11:
@mirsadm Do I need to any change to logcat options or is it the same as in the article?