mirsadm/motioncam #48: Build error

dimka11, 2021-12-09:
I have tried build project on MacOS in Android Studio, but get error:

```
Build command failed.
Error while executing process /Volumes/MACOSAPPS/Android/SDK/cmake/3.10.2.4988404/bin/ninja with arguments {-C /Volumes/MACOSAPPS/motioncam/MotionCam-Android/MotionCam/.cxx/Debug/3075m2v5/arm64-v8a native-camera-host native-camera-opencl}
ninja: Entering directory `/Volumes/MACOSAPPS/motioncam/MotionCam-Android/MotionCam/.cxx/Debug/3075m2v5/arm64-v8a'

ninja: error: '../../../../../../libMotionCam/libMotionCam/halide/arm64-v8a/halide_runtime_host.a', needed by '../../../../build/intermediates/cxx/Debug/3075m2v5/obj/arm64-v8a/libnative-camera-host.so', missing and no known rule to make it
```

Any ideas?

mirsadm, 2021-12-09:
Have you run setupenv.sh?

dimka11, 2021-12-10:
Yes, but probably it's not install all things, like I don't see holide folder in third-party folder

mirsadm, 2021-12-13:
That script will need to run before you can build the Android app. Let me know if there's a problem with the script for you.

dimka11, 2021-12-14:
@mirsadm I ran script before

mirsadm, 2021-12-23:
Did it complete without failing?