mirsadm/motioncam #50: Camera preview freeze after couple of seconds

ishansd94, 2021-12-11:
I have tried the latest APK on my S21 Ultra Snapdragon. After I open the app, the camera preview freeze but I can access all the other settings. Any idea?

ishansd94, 2021-12-11:
@mirsadm found this logs

```
12-11 15:07:42.999 18996 18996 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
12-11 15:07:42.999 18996 18996 F DEBUG   : Build fingerprint: 'samsung/p3quew/p3q:12/SP1A.210812.016/G998U1UEU4BUKF:user/release-keys'
12-11 15:07:42.999 18996 18996 F DEBUG   : Revision: '18'
12-11 15:07:42.999 18996 18996 F DEBUG   : ABI: 'arm64'
12-11 15:07:42.999 18996 18996 F DEBUG   : Timestamp: 2021-12-11 15:07:42.797219946+0530
12-11 15:07:42.999 18996 18996 F DEBUG   : Process uptime: 2s
12-11 15:07:42.999 18996 18996 F DEBUG   : Cmdline: com.motioncam
12-11 15:07:42.999 18996 18996 F DEBUG   : pid: 18818, tid: 18992, name: com.motioncam  >>> com.motioncam <<<
12-11 15:07:42.999 18996 18996 F DEBUG   : uid: 10009
12-11 15:07:42.999 18996 18996 F DEBUG   : signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
12-11 15:07:42.999 18996 18996 F DEBUG   :     x0  0000000000000000  x1  0000000000004a30  x2  0000000000000006  x3  0000007a13d44940
12-11 15:07:42.999 18996 18996 F DEBUG   :     x4  466b621f39636d74  x5  466b621f39636d74  x6  466b621f39636d74  x7  7f7f7f7f7f7f7f7f
12-11 15:07:42.999 18996 18996 F DEBUG   :     x8  00000000000000f0  x9  5e57100f3f350ba6  x10 0000000000000000  x11 ffffff80fffffbdf
12-11 15:07:42.999 18996 18996 F DEBUG   :     x12 0000000000000001  x13 000000000000002f  x14 0000000000000000  x15 0000000034155555
12-11 15:07:42.999 18996 18996 F DEBUG   :     x16 0000007da74db060  x17 0000007da74b7a80  x18 0000007a12d3a000  x19 0000000000004982
12-11 15:07:42.999 18996 18996 F DEBUG   :     x20 0000000000004a30  x21 00000000ffffffff  x22 0000007bb20f8fdf  x23 0000007a2d2a0820
12-11 15:07:42.999 18996 18996 F DEBUG   :     x24 0000007a13d45cb0  x25 0000007a13d45cb0  x26 0000007a13d45ff8  x27 0000000000000000
12-11 15:07:42.999 18996 18996 F DEBUG   :     x28 0000007a13c4d000  x29 0000007a13d449c0
12-11 15:07:42.999 18996 18996 F DEBUG   :     lr  0000007da7467bf4  sp  0000007a13d44920  pc  0000007da7467c20  pst 0000000000001000
12-11 15:07:42.999 18996 18996 F DEBUG   : backtrace:
12-11 15:07:42.999 18996 18996 F DEBUG   :       #00 pc 0000000000051c20  /apex/com.android.runtime/lib64/bionic/libc.so (abort+168) (BuildId: ce9d8892515aab3292de60cde842627f)
12-11 15:07:42.999 18996 18996 F DEBUG   :       #01 pc 000000000058db5c  /data/app/~~AT7qsbYvkLr2VrFqTl1BHw==/com.motioncam-TrgNYGF3O451jnwJJ0nImw==/split_config.arm64_v8a.apk!libnative-camera-opencl.so (halide_default_error+116) (BuildId: be4b2b2f543f899008f0eabedf1c61c343499020)
12-11 15:07:44.440 19105 19284 F libc    : Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 19284 (com.motioncam), pid 19105 (com.motioncam)
12-11 15:07:44.704 19288 19288 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
```

mirsadm, 2021-12-11:
Ah yep that's great. Are you able to get more of the logs for me by any chance?

ishansd94, 2021-12-11:
I can get you logs.  I just have couple of questions
1. I used this command with options `.\adb.exe -d logcat com.motioncam:E > motioncam.log`. Is there any other options I should pass in?
2. Is there a way to enable debug logs in the app it self?

mirsadm, 2021-12-11:
I've put up a beta release here: https://github.com/mirsadm/motioncam/releases/download/7.0.8-beta/MotionCam.7.0.8-beta.apk

Could you test to see if that fixes the issue?

ishansd94, 2021-12-11:
@mirsadm same issue is happening on the beta as well

ishansd94, 2021-12-11:
@mirsadm the complete log file https://ufile.io/v3q2f58z

mirsadm, 2021-12-11:
Thanks for the logs. I have them downloaded so feel free to remove them in case they have some sensitive information.

Kidwithservers, 2021-12-13:
I just wanted to note that I am having the same issue on the same phone, with the beta.

mirsadm, 2021-12-13:
This is a long shot but could you try it again here:
https://github.com/mirsadm/motioncam/releases/download/7.0.8-beta2/MotionCam.7.0.8-beta2.apk

I'm taking guesses as there's no obvious cause for this.

ishansd94, 2021-12-14:
@Kidwithservers any luck?
@mirsadm I still facing the same issue. Could it be related to Android 12?

ishansd94, 2021-12-14:
i have noticed couple of warning related to camera activity process in Android 12. Could it be the reason?

```
12-14 14:47:59.573 16481 16541 I WidgetsPredictionUpdateTask: 1. Predicted target com.motioncam#UserHandle{0}
12-14 14:48:00.070 16481 16541 I WidgetsPredictionUpdateTask: 1. Predicted target com.motioncam#UserHandle{0}
12-14 14:48:16.084 15111 15750 I Pageboost: IO Prefetch for : com.motioncam
12-14 14:48:16.086  4018  8005 E pageboostd: prefetch start : app commotioncam
12-14 14:48:16.089  4018  8005 E pageboostd: commotioncam, amt 1155072 scnt 7 fcnt 4
12-14 14:48:16.089  4018  8005 E pageboostd: prefetch end : app commotioncam data_amount 1155072
12-14 14:48:16.588 16481 16481 W AppStartUtils: sendActiveLaunchBroadcast() v.hasOnClickListeners() : true, iconInfo : AppInfo(id=644 type=APP hidden=0 container=all_apps screen=6 rank=3 rankOfList=-1 user=UserHandle{0} title=Motion Cam componentName=ComponentInfo{com.motioncam/com.motioncam.CameraActivity} screenType=0)
12-14 14:48:16.589 15111 15742 I Pageboost: start alp : com.motioncam
12-14 14:48:16.589 15111 15750 I Pageboost: alp for : com.motioncam , 1
12-14 14:48:16.590  4018  8012 E pageboostd: alp start : app commotioncam
12-14 14:48:16.591  4018  8012 E pageboostd: commotioncam, amt 13383600 scnt 2 fcnt 0
12-14 14:48:16.591  4018  8012 E pageboostd: commotioncam, amt 1155072 scnt 7 fcnt 4
12-14 14:48:16.591  4018  8012 E pageboostd: alp end : app commotioncam data_amount 14538672
12-14 14:48:16.657 16481 16541 I AppIconSolution: getThemeIconWithBG called with public API, pkg = com.motioncam
12-14 14:48:16.658 16481 16541 I AppIconSolution: start to load, pkg=com.motioncam, bg=192-192, dr=192-192, forDefault=true, density=640
12-14 14:48:16.658 16481 16541 I AppIconSolution: getIconScale, pkg=com.motioncam, size=192, iconScale=IconScale[alpha=0, scale=0.72, isCrop=false]
12-14 14:48:16.659 15111 16511 I ActivityTaskManager: START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10200000 cmp=com.motioncam/.CameraActivity bnds=[1060,356][1388,771]} from uid 10153
12-14 14:48:16.661 16481 16541 I AppIconSolution: default container[Contain], pkg=com.motioncam, bg=266-266, dr=192-192, isNight = true
12-14 14:48:16.662 15111 16511 D FreecessController: UFZ : com.motioncam(28510)
12-14 14:48:16.662 16481 16541 I LauncherActivityInfo: packageName: com.motioncam, useThemeIcon: false, height: 266, width: 266, density: 640
12-14 14:48:16.662 15111 16511 D FreecessController: UFZ : com.motioncam(0), reason: activity
12-14 14:48:16.663 15111 16511 D BaseRestrictionMgr: Package: com.motioncam, userid: 0, hostingType: activity is allowed by freecess, caller is: com.sec.android.app.launcher
12-14 14:48:16.663 15111 16511 D ActivityTaskManager: TaskLaunchParamsModifier:task=null activity=ActivityRecord{b01002e u0 com.motioncam/.CameraActivity display-area-from-source=DefaultTaskDisplayArea@260336953 display-id=0 display-windowing-mode=1 suggested-display-area=DefaultTaskDisplayArea@260336953
12-14 14:48:16.663 15111 16511 D [secipm]: mSecIpmManager Preload com.motioncam dex files
12-14 14:48:16.663 15111 16511 D ActivityTaskManager: TaskLaunchParamsModifier:task=null activity=ActivityRecord{b01002e u0 com.motioncam/.CameraActivity t-1} display-area-from-source=DefaultTaskDisplayArea@260336953 display-id=0 display-windowing-mode=1 suggested-display-area=DefaultTaskDisplayArea@260336953 inherit-from-source=fullscreen activity-options-fullscreen=Rect(0, 0 - 0, 0) non-freeform-display display-area=DefaultTaskDisplayArea@260336953 maximized-bounds
12-14 14:48:16.663 15111 17123 D PkgPredictorService-IpmAdcpController: create a new DexPreloadTask pkg:com.motioncam  path:/data/app/~~41UEuVZAeWoVJ2slYzSypg==/com.motioncam-MfQXFdg_XpmCD5FdpfkDlw==
12-14 14:48:16.665 15111  8018 D PkgPredictorService-IpmAdcpController: /data/app/~~41UEuVZAeWoVJ2slYzSypg==/com.motioncam-MfQXFdg_XpmCD5FdpfkDlw==/oat/arm64/base.art is already preloaded
12-14 14:48:16.665 15111 16511 D ActivityTaskManager: TaskLaunchParamsModifier:task=Task{a7cd55c #4578 type=standard A=10814:com.motioncam U=0 visible=false mode=fullscreen translucent=true sz=0} activity=ActivityRecord{b01002e u0 com.motioncam/.CameraActivity t-1} display-area-from-source=DefaultTaskDisplayArea@260336953 display-id=0 display-windowing-mode=1 suggested-display-area=DefaultTaskDisplayArea@260336953 inherit-from-source=fullscreen activity-options-fullscreen=Rect(0, 0 - 0, 0) non-freeform-display display-area=DefaultTaskDisplayArea@260336953 maximized-bounds
12-14 14:48:16.667  1177  3767 I SurfaceFlinger: id=48246 createSurf (0x0),-1 flag=80004, ActivityRecord{b01002e u0 com.motioncam/.CameraActivity t4578}#0
12-14 14:48:16.670 15111 16511 D MARsPolicyManager: onPackageResumedFG pkgName = com.motioncam, userId = 0
12-14 14:48:16.673 15111 15111 I Pageboost: package com.motioncam
12-14 14:48:16.673 15111 16517 D MARsPolicyManager: onPackageResumedFG pkgName = com.motioncam, userId = 0
12-14 14:48:16.675 15111 16517 D WindowManager: rotationForOrientation, orientationSource=ActivityRecord{b01002e u0 com.motioncam/.CameraActivity t4578}
12-14 14:48:16.680 15111 16517 D SGM:PkgDataHelper: notifyAppCreate(), pkgName: com.motioncam, userId: 0, sendRet: true
12-14 14:48:16.680 15111 15539 D SGM:GameManager: onLooperPrepared(), msg: MSG_APP_CREATE, pkgName: com.motioncam, userId: 0
[... cut ...]
12-14 14:48:18.073 15111 15111 D DualDARComnService: reConnectAgentsByPkgName:: package: com.motioncam
12-14 14:48:18.202 16481 16541 I WidgetsPredictionUpdateTask: 1. Predicted target com.motioncam#UserHandle{0}
12-14 14:48:18.338  5644  5903 I SAMSUNGPAY: [Hint_sg2] package changed : com.motioncam , execute : true
12-14 14:48:18.342 16481 16541 I PackageUpdatedTask: mAllAppsList.updatePackage com.motioncam
12-14 14:48:18.348 16481 16541 I AppIconSolution: getThemeIconWithBG called with public API, pkg = com.motioncam
12-14 14:48:1
```

ishansd94, 2021-12-14:
@mirsadm I was able to test the new beta4 and found a way to isolate logs from the PID.  Here are the logs https://ufile.io/5jguk2b2 

Hope it helps.

Kidwithservers, 2021-12-14:
No luck with the new update. I am also on android 11 currently.

mirsadm, 2021-12-15:
I think what I'll need to do is disable everything but the basics and see if that works. It looks like the camera service is crashing.

ishansd94, 2021-12-15:
@mirsadm if you can do a build with the basics I can test and send you the logs. The same camera preview freeze is happening on some of the GCam mods as well and the error was `Camera HAL package was not found`. I don't know if this is related or relevant to this issue.

danny-ray, 2021-12-15:
@mirsadm 
First of all, thumbs up for that app, it's a really nice idea.

But i've got exactly the same "freeze problem" like ishansd94.
It takes 3 seconds before it freezes. But when you switch the cam or go to the settings forth and back, it always resets
for another 3 seconds.

In my case it's a Samsung Galaxy Zflip 3 5G with android 11 and newest beta.
I also tested it on a Samsung Galaxy S7 tablet with version 7.0.3 and it worked flawlessly.

Hope you will find the problem.

mirsadm, 2021-12-15:
I've made a release here for testing if anyone wants to give it a go:
https://www.file.io/download/Zi0sfESIqovC

It disabled lots of stuff. Hoping that fixes it.

ishansd94, 2021-12-16:
@mirsadm file is deleted.

ishansd94, 2021-12-17:
@mirsadm looks the file is deleted. Can you upload it again?

mirsadm, 2021-12-17:
Ah yeah I'll put something up on GitHub instead

mirsadm, 2021-12-17:
Here we go:
https://github.com/mirsadm/motioncam/releases/download/7.0.8-beta5/MotionCam-Samsung-Test.apk

ishansd94, 2021-12-18:
> Here we go: https://github.com/mirsadm/motioncam/releases/download/7.0.8-beta5/MotionCam-Samsung-Test.apk

@mirsadm Same error. Could it be related to package name? I've seen that GCams use a separate package name for Samsung devices. This way the app can access all the 4 lenses. Worth a try perhaps?

```
com.samsung.android.ruler
com.samsung.android.scan3d
com.samsung.android.biometrics.service
```

mirsadm, 2021-12-18:
It sounds like the Samsung camera service is crashing. It would be worthwhile asking devs that have dealt with it with the gcam ports. 

No I'm not but I think I'll start a group on there. It would be helpful for stuff like this.

ishansd94, 2021-12-18:
> Here we go: https://github.com/mirsadm/motioncam/releases/download/7.0.8-beta5/MotionCam-Samsung-Test.apk

@mirsadm Here are the full device logs: https://file.io/T98kLVztZ4IU 

Notice the lines from 8650

```
12-18 15:30:26.662 11953 16846 W CHIUSECASE: [WARN   ] chifeature2graphmanager.cpp:345 LoadFeatureGraphSelectorOps() No graph selector library files found in the path /vendor/lib64/camera/components/feature2/oem
12-18 15:30:26.663  4145  4145 I VSET/SettingsActivity: ActivateSettingActivityForSupportedCases: country true
12-18 15:30:26.664  4145  4145 I VSET/CameraUtil: getUnihalVersion: 2.0
12-18 15:30:26.665 11953  9625 E CamX    : [ERROR][ISP    ] secgtmclass.h:32 CGtmDrv() [gtmdrv] create this = 0x79c1d85250
12-18 15:30:26.666 11953 16846 W CHIUSECASE: [WARN   ] chifeature2hwmultiframe.cpp:2955 DoQueryCaps() pEnableMFFeature in SessionSetting is NULL
12-18 15:30:26.669 11953 16846 I CHIUSECASE: [CONFIG ] chithreadmanager.cpp:28 Create() FeatureThreadManager: NumThreads: 50
12-18 15:30:26.670 15111 16511 D PackageManager: setEnabledSetting : userId = 0 packageName = com.samsung.android.vtcamerasettings cmp = com.samsung.android.vtcamerasettings.settings.SettingsActivity newState = 1 callingPackage = 1000/com.samsung.android.vtcamerasettings
12-18 15:30:26.671 11953 16846 W CHIUSECASE: [WARN   ] chifeature2generic.cpp:1127 OnPreparePipelineCreate() pPipelineData 0x7931a96c60 or pipeline is NULL
12-18 15:30:26.671  4145  4145 I VSET/CameraUtil: getUnihalVersion: 2.0
12-18 15:30:26.672 11953 16846 E CamX    : [ERROR][CORE   ] camxtitan17xcontext.h:248 GetHwVersion() Platform version 5.8.0, cpas version 1.0.0, soc id 415
12-18 15:30:26.672 11953 16846 E CHIUSECASE: [ERROR  ] chifeature2hwmultiframe.cpp:2704 OnInitializeStream() PortName:MFSR_Postfilter_YUV_OUT_2 Res = 4000X3000 format = 35
12-18 15:30:26.673  4145  4145 I VSET/ObserveCameraStateService: onStartCommand: update SALogging status
12-18 15:30:26.673 11953 16846 I CHIGPU  : camxchinodegpu.cpp:4290 Initialize(): bypass set 0, inplace set 0
12-18 15:30:26.673 11953 16846 I CHIGPU  : camxchinodegpu.cpp:4317 Initialize(): Setprop to disable bufferMapping 0 disableSIMODS 0
12-18 15:30:26.676  4145  4145 I VSET/CameraStateManagerImpl: updatePackageInfo: notify update
12-18 15:30:26.676 11953 16846 I CHIGPU  : camxchinodegpu.cpp:4342 Initialize(): StreamTypePresent
12-18 15:30:26.676  4145  4145 I VSET/CameraStateManagerImpl: updatePackageInfo: CameraStateInfo id= 0(CLOSED), packageName = com.motioncam,
12-18 15:30:26.676  4145  4145 I VSET/CameraStateManagerImpl:  supportedList -> SETTINGS_AUTO_FRAMING_MODE_MENU = false,SETTINGS_AUDIO_CONFIGURATION_MENU = false
12-18 15:30:26.677  4145  4145 I VSET/ObserveCameraStateService: start camera : get intent : com.motioncam, 0
12-18 15:30:26.677  4145  4145 I VSET/ObserveCameraStateService: updateAppInfoRepository: E 
12-18 15:30:26.677  4145  4145 I VSET/ObserveCameraStateService: initAppRepository is created already
12-18 15:30:26.678 11953 16846 I CHIGPU  : camxchinodegpu.cpp:4450 QueryBufferInfo(): numOutputPorts = 1, numInputPorts = 1
12-18 15:30:26.678 11953 16846 I CHIGPU  : camxchinodegpu.cpp:4648 QueryBufferInfo(): Input port id = 0, Format = 3  m_nodeCaps: 32 max 4000x3000 min 4000x3000
12-18 15:30:26.678 11953 16846 I CHIUSECASE: [CONFIG ] chifeature2base.cpp:8669 CreatePipeline() MFSR [Blend_UBWC_OUT] Providing buffer negotiation options(0x78e1b9c820) to{65538, 1, 10} on stream(0x7931acb060) min [4000 x 3000] max [16383 x 16383] optimal [4000 x 3000] isFrameworkSink: 0
12-18 15:30:26.678 11953 16846 I CHIUSECASE: [CONFIG ] chifeature2base.cpp:8669 CreatePipeline() MFSR [Blend_PD10DS4_OUT] Providing buffer negotiation options(0x78e1be0da0) to{65538, 1, 11} on stream(0x7931a7f190) min [4000 x 3000] max [16383 x 16383] optimal [4000 x 3000] isFrameworkSink: 0
12-18 15:30:26.679 11953 16846 E CamX    : [ERROR][LRME/CVP] camxcvpnode.h:2070 SetTitanVersion()  setting 329728
12-18 15:30:26.680 11953 16846 E CamX    : [ERROR][CORE   ] camxtitan17xcontext.h:248 GetHwVersion() Platform version 5.8.0, cpas version 1.0.0, soc id 415
12-18 15:30:26.680 11953 16846 I CHIUSECASE: [CONFIG ] chifeature2base.cpp:8669 CreatePipeline() MFSR [Prefilter_UBWC_OUT] Providing buffer negotiation options(0x78e1b9c820) to{65538, 0, 10} on stream(0x7931ab5a00) min [640 x 480] max [16383 x 16383] optimal [4000 x 3000] isFrameworkSink: 0
12-18 15:30:26.680 11953 16846 I CHIUSECASE: [CONFIG ] chifeature2base.cpp:8669 CreatePipeline() MFSR [Prefilter_PD10DS4_OUT] Providing buffer negotiation options(0x78e1bd9120) to{65538, 0, 11} on stream(0x7931ad2ac0) min [640 x 480] max [16383 x 16383] optimal [4000 x 3000] isFrameworkSink: 0
12-18 15:30:26.680 11953 16846 I CHIUSECASE: [CONFIG ] chifeature2base.cpp:8669 CreatePipeline() MFSR [Prefilter_PD10DS16_OUT] Providing buffer negotiation options(0x78e1be7560) to{65538, 0, 12} on stream(0x7931aae310) min [640 x 480] max [16383 x 16383] optimal [4000 x 3000] isFrameworkSink: 0
12-18 15:30:26.680 11953 16846 I CHIUSECASE: [CONFIG ] chifeature2base.cpp:8669 CreatePipeline() MFSR [Prefilter_PD10DS64_OUT] Providing buffer negotiation options(0x78e1bc69e0) to{65538, 0, 13} on stream(0x7931ac5860) min [640 x 480] max [16383 x 16383] optimal [4000 x 3000] isFrameworkSink: 0
12-18 15:30:26.681 11953 16846 E CamX    : [ERROR][LRME/CVP] camxcvpnode.h:2070 SetTitanVersion()  setting 329728
[... cut ...]
12-18 15:30:26.866 14908 22211 I Camera3-Status: addComponent: Adding new component 3 - streamId = 1, format = 37, width = 4000, height = 3000, option = 0
12-18 15:30:26.866 14908 22211 D Camera3-OutputStream: configureConsumerQueueLocked: Consumer wants 4 buffers, HAL wants 8
12-18 15:30:26.867 14908 22211 D Camera3-Device: CameraPerf: Connect Streams X
12-18 15:30:26.867 14908 22211 D Camera3-Device: Set real time priority for request queue thread (tid 9616)
12-18 15:30:26.867 14908 22211 D Camera3-Device: CameraPerf: configureStreamsLocked: Camera 0: Stream configuration complete
```

ishansd94, 2021-12-18:
@mirsadm I did a little digging and found out these errors.

```
12-18 15:30:57.874 11953 10745 E libc    : Access denied finding property "persist.camera.af.lib.debug"
12-18 15:30:57.874 11953 10745 E libc    : Access denied finding property "persist.camera.af.lib.osd"
```

also there are multiple `avc:  denied` errors

```
12-18 15:30:56.964   926   926 E audit   : type=1400 audit(1639821656.960:103954): avc:  denied  { read } for  pid=11953 comm="provider@4.0-se" name="u:object_r:default_prop:s0" dev="tmpfs" ino=14448 scontext=u:r:hal_camera_default:s0 tcontext=u:object_r:default_prop:s0 tclass=file permissive=0 SEPF_SM-G998U1_11_0010 audit_filtered
12-18 15:30:56.964   926   926 E audit   : type=1300 audit(1639821656.960:103954): arch=c00000b7 syscall=56 success=no exit=-13 a0=ffffff9c a1=786e3d1af8 a2=88000 a3=0 items=1 ppid=1 pid=11953 auid=4294967295 uid=1047 gid=1005 euid=1047 suid=1047 fsuid=1047 egid=1005 sgid=1005 fsgid=1005 tty=(none) ses=4294967295 comm="provider@4.0-se" exe="/vendor/bin/hw/vendor.samsung.hardware.camera.provider@4.0-service_64" subj=u:r:hal_camera_default:s0 key=(null)
```

ishansd94, 2021-12-21:
@mirsadm let me know if you need anything else

ishansd94, 2021-12-21:
@mirsadm logs for 8.0beta2

By PID: https://ufile.io/aawd9tly
Full Device logs: https://ufile.io/3p18fwxd

mirsadm, 2021-12-21:
Thanks for the logs but I'm not sure this is something I can fix. After some research it seems to be a problem with that particular phone. If you are able to get more information from one of the developers working on a GCam port for that phone they may be able to help

mirsadm, 2021-12-22:
Can someone try this release:
https://github.com/mirsadm/motioncam/releases/download/8.0.0-beta3/MotionCam-Samsung-Test.apk

ishansd94, 2021-12-23:
@mirsadm no luck. facing the same issue.

ishansd94, 2021-12-23:
@mirsadm is it possible to change the package name to `com.samsung.android.ruler` for the samsung build?

mirsadm, 2021-12-26:
Could you try this one:
https://github.com/mirsadm/motioncam/releases/download/8.0.4/MotionCam.8.0.4.apk

I've changed a couple of things again

ishansd94, 2021-12-27:
@mirsadm still having the same issue.

One of the Gcam devs got in touch and said to try this out.
```
The thing you're looking for is Qualcomm Stream Configs. It needs to be set to 61446.
```

Do you have an idea?

ishansd94, 2021-12-27:
> Could you try this one: https://github.com/mirsadm/motioncam/releases/download/8.0.4/MotionCam.8.0.4.apk
> 
> I've changed a couple of things again

logs: https://ufile.io/tezxqnsf

mirsadm, 2021-12-30:
Still looking into this, I'll contact the developer

ishansd94, 2021-12-30:
Thank you @mirsadm.

8rnity, 2022-01-02:
hello, Samsung Galaxy Zflip 3 with Snapdragon has the same issue

cnarco, 2022-01-04:
Yeah same with the A52s 5G... I used motioncam on my oneplus before but I broke the screen so now I'm struggling with samsung...btw I can't manage to find a working Gcam version and I'm on android 11 so idk

mirsadm, 2022-01-07:
Still looking into fixing tbis, very annoying!

cnarco, 2022-01-07:
these back cam settings fixed my gcam so might be good for this one also. 
[image]

danny-ray, 2022-01-07:
> Still looking into fixing this, very annoying!
Sounds good.

Had updated my Samsung Z Flip 3 5G to UI4 and Android 12(which  included several cam updates)
but it still doesn't work proper...

Jeneeeeek, 2022-01-09:
> Все еще ищу возможность исправить это, очень раздражает!

there are contacts of many developers of mod of gcam, if you need to help add sessions, they will help

ishansd94, 2022-01-10:
@mirsadm it's actually possible to capture a 1 second video until the freeze.

https://streamable.com/eeb4yj

ishansd94, 2022-01-10:
@mirsadm it's actually possible to capture a 1 second video until the freeze.

https://user-images.githubusercontent.com/12171913/148726036-ce5e429e-23c1-4e76-83df-301f146d19a6.mp4