mirsadm/motioncam #51: [UI tweaks] Crop lines in raw video mode and focus slider adjustment

slonick81, 2021-12-14:
1) Is it possible to change 1/3 photo guidelines into crop marks in video mode for better framing experience? If I'm not wrong it's in the end of this file:
motioncam/MotionCam-Android/MotionCam/src/main/java/com/motioncam/ui/CameraGridLayout.java
lines 60-64:
`        canvas.drawLine(width/3, 0, width/3, height, mPaint);
        canvas.drawLine(width/3*2, 0, width/3*2, height, mPaint);

        canvas.drawLine(0, height/3, width, height/3, mPaint);
        canvas.drawLine(0, height/3*2, width, height/3*2, mPaint);`
Is it possible to get mode check and if it's video change "width/3" and "width/3*2" to something like "width*vert_crop" and "width-width*vert_crop", and so on for height?

2) Is it possible to get different focus slider mapping? It's much more usable now but nearest meter still gets half of slider scale. I understand that it may be phone specific and different users may have different needs so maybe it's possible to add some log scale remapping in settings to adjust it? Also is it possible to add some horizontal space/size to focus slider and hide it on second tap on focus distance, not automatically right after adjustment? It won't make any difference for AF shooting but will make manual focus much more usable. I have rudimentary programming skills so if you poke me into code responsible for this I may give more specific suggestions.

Thank you for great work anyway!

2blackbar, 2021-12-18:
i like the idea of displaying crop marks for a few seconds whwn adjusting crop, or of course displaying marks all the time if possible

slonick81, 2021-12-18:
Yes. There is so little point in 1/3 marks in crop mode anyway cause you're capturing different screen area - it's cropped (and you can't see how). They are 1/3rds no more.

mirsadm, 2021-12-20:
I've changed the video mode to show the cropped video. You can try it out:
https://github.com/mirsadm/motioncam/releases/download/8.0.0-beta0/MotionCam.8.0.0-beta0.apk

slonick81, 2021-12-20:
Thank you! Seems to work, but axis look swapped. In settings V cropping leads to vertical cropping in landscape mode (in recorded files at least) but cache is added on left and right sides, like if it was portrait mode.

2blackbar, 2021-12-20:
YES works fantastic but swapped, that should be easy fix.

mirsadm, 2021-12-20:
Yep fixed!
https://github.com/mirsadm/motioncam/releases/download/8.0.0-beta1/MotionCam.8.0.0-beta1.apk

slonick81, 2021-12-20:
Works fine, confirmed

closed 2021-12-23 by mirsadm