mirsadm/motioncam #53: pinch zoom, crop mask, few ideas

cnarco, 2021-12-17:
-add version info to settings
-pinch zoom would be great or at least a manual zoom slider in camera mode
-manual focus slider for video recording
-visualization for cropped areas in video mode
-zebra mode
-I tried to record raw it's awesome but unfortunately there's massive amount of dropped frames for me (oneplus 7t 8gb ram). I've tried to ramp up the memory usage in settings even lowered the frame rates but can't manage to record usable footage

(really cool project congrats)

mirsadm, 2021-12-17:
I've made some progress on improving performance. I'll put out an update soon. There is a non obvious way to set manual focus and change the iso/shutter speed. If you click on the focus distance/iso/shutter speed you can change it. 
Thanks for the suggestions

mirsadm, 2021-12-20:
I'm going to close this issue for now. This new release addresses some of your issues:
https://github.com/mirsadm/motioncam/releases/download/8.0.0-beta0/MotionCam.8.0.0-beta0.apk

closed 2021-12-20 by mirsadm