mirsadm/motioncam #55: Feature idea: tensorflow based deblurring etc.

joemarshall, 2021-12-17:
I have been playing a bit with deep learning based deblurs, super resolution etc. 

It would be great to have some of these integrated into motioncam in addition to the denoising stuff.

If it was restricted to fully convolutional models that could be built as tensorflow lite models then it would be relatively simple technically because tensorflow support on Android is pretty trivial, and fully convolutional models can work for any input shape.

joemarshall, 2021-12-17:
As an example I've just coded the model from this paper
https://www.researchgate.net/publication/339977282_Multi-Scale_Neural_Network_With_Dilated_Convolutions_for_Image_Deblurring

And am training it on the realblur dataset, it seems it has pretty good results for static scenes with camera shake at least.

mirsadm, 2021-12-17:
That sounds very interesting and I'm all up for trying it out. Are you able to produce a proof of concept using the app? I'll be happy to help you out if you need assistance