mirsadm/motioncam #57: List bugs on OnePlus 8T

kola4k, 2021-12-19:
All those who have bugs on this smartphone model can add to the list:
1) When switching to other cameras, the black screen shows.
2) But if you take a picture, then a picture appears. Moreover, this method works for all cameras except one. If you take a picture on the third camera, the application crashes.
3) On the front camera, everything is the same. Another inverted image!!! But the correct orientation is maintained. When the photo is not taken, a black screen is displayed. But the camera power-on indication is working and the numbers in the lower left corner are changing.
4) Screen flip works even if it is disabled in the system. Inconvenient.

kola4k, 2021-12-23:
In the new version 8.0.0-beta 3, most of the bugs related to the inclusion of cameras have disappeared, but one camera still does not start, but the application does not crash at the same time. 
But there are new bugs. For example, when photographing, some kind of thin frame is added to the sides of the photo, similar to a strip of broken squeaks. This is evident both on the main and on the front cameras.

kola4k, 2021-12-23:
[image]
[image]

kola4k, 2021-12-23:
In BURST mode, the bug also manifests itself. After processing the photo.

kola4k, 2021-12-23:
I would also like the following functions in the future. Maximum shooting speed mode, maximum frames per second. And record stereo sound from an external microphone. As well as saving the footage to an external SSD drive. Then it will be possible to make a full-fledged tool for shooting from a smartphone. Thanks for the work, I will send a donation in the future :)

mirsadm, 2021-12-23:
Ah yeah I can reproduce that on my phone. FYI you can already record to an external drive by changing the location in the settings.

mirsadm, 2021-12-23:
That's been fixed in the latest build

closed 2021-12-23 by mirsadm