mirsadm/motioncam #61: Failed DNG conversion of RAW video in 8.0.0-beta0

slonick81, 2021-12-20:
When I try to convert raw video into DNG sequence in "manage video" menu it changes the button to "queued" but progress bar immediately disappears and nothing else happens. It creates empty folder with proper name and attempt counter for each try, and if I re-enter into this conversion menu it adds "This video has been exported" sign to the clip. The clip looks OK - proper capture time, fps, frame count and preview strip.
Poco X3 6G/128 NFC

slonick81, 2021-12-20:
Tried clean setup - uninstalled the app, emptied the destination folder for dng converter, ran system cleanup. Intalled the app, granted all permissions, set up dng folder with permission, too. Still no success but the symptoms are different: I'm getting message "The selected output folder is unavailable. Using internal storage." twice at the beginning of recording. It's still impossible to convert clip to dng but now I can see zero-sized .container and .wav files in addition to empty conversion folder in the destination path. I guess now it tries to create audio and image files but fails for some reason.

2blackbar, 2021-12-20:
it sometimes locks up when you tap too many times on the same footage to convert, just do it once and wait.

mirsadm, 2021-12-20:
@slonick81 is there free space where you're trying to convert to?

hoodFX, 2021-12-20:
i also experienced this issue, until you release fixed the issue at 8.0.0 beta1 version.

slonick81, 2021-12-21:
Made some tests this night (on 8.0.0 beta1). Tried different stuff like rebooting the phone, reinstalling the app, manual permission settings, different resolutions and framerates and so on. There is enough storage space (70GB of 128 free). I discovered one pattern that reliably lets me to convert raw video:
1. make current dng conversion path invalid (like delete/move/rename folder in external file browser)
2. shoot new clip
3. go to "manage videos", tap button to queue
4. app will ask to set new directory for conversion - set it
5. the conversion begins
If conversion process is interrupted by going outside the menu to main screen or due to phone locking itself then things get unpredictable. I have several tries when I was able to requeue the conversion and complete it but mostly it hangs instantly or (rarely) it converts a small portion of clip and hangs.
I hope it may give some ideas. Maybe there is a log of some kind I can extract and send?

2blackbar, 2021-12-21:
Yeah thats pretty much what caused these thngs on my phone but now i dont delete and change folders around much so it doesnt happen, also i dont tap twice to convert to dng, when it says "this clip was exported" i just delete it, it would be nice to have an option in settings that would auto delete after conversion to save space.

mirsadm, 2021-12-23:
I've reworked that screen significantly, it'll hopefully reduce/eliminate a lot of these issues. Let me know how you go with the latest release.

2blackbar, 2021-12-26:
It works great, i wish big companies listened to users like that and  acred about these time saving functions in their software, it really makes the whole experience so much better.
I also recommend something like "export all to dng" so we dont have to click one by one on the list to export the containers to dngs....

slonick81, 2021-12-27:
Well, still same problems on 8.0.3. I've tested microSD card as export destination, made factory reset for the phone (in case some side app was messing with storage access) but no luck so far. I can not write temp .container and .wav to any directory other then default (files are created but 0 byte size), I need to set "fresh" folder at queueing button tap (as I wrote earlier). BTW it's possible to queue multiple older files with this hack, not only the recent one. It's somewhat usable as it is predictable and I don't loose shots at the end but it's far from being straight.

mirsadm, 2021-12-27:
What Android version and phone are you using? There's nothing unusual about the way I write to those folders.

slonick81, 2021-12-27:
Android 10 QKQ1.200512.002

mirsadm, 2022-01-04:
Not sure what's going on with this one but if you are able to get the files of your phone you can use the Windows/Mac tools now to convert

mirsadm, 2022-01-11:
Closing this for now

closed 2022-01-11 by mirsadm