mirsadm/motioncam #63: 8.0.0-beta0 issues

miestopaukstis, 2021-12-20:
Hi, thanks for big update, here's few issues i've noticed using Mi 11 Ultra:
- OIS toggle dissapeared
- Crop preview tool looks very nice and welcomed, but somehow it seems to be working (or displaying) opposite
- Recorded RAW video are placed not in zip files, but looks like filetype is recoginized like a VIDEOxxxx.container . So external conversion seems not possible with tools that I do have from previous version.

2blackbar, 2021-12-20:
you tried unzip it ?

mirsadm, 2021-12-20:
ZIPs are gone. Unfortunately they were adding too much overhead so its using a custom file format now. It provides a pretty significant boost in write speed.

mirsadm, 2021-12-20:
Crop is fixed here:
https://github.com/mirsadm/motioncam/releases/download/8.0.0-beta1/MotionCam.8.0.0-beta1.apk

closed 2021-12-20 by mirsadm