mirsadm/motioncam #65: Crop menu not show in 8.0.0 beta1

hoodFX, 2021-12-20:
From v 7.0.8 to 8.0.0 beta 1, i can't access crop menu. when i tap crop menu, there's no slider option to choose dimension for video recording. I used HTC U11.

mirsadm, 2021-12-20:
Possibly not enough space, I've updated it to hide the manage videos button
https://github.com/mirsadm/motioncam/releases/download/8.0.0-beta2/MotionCam.8.0.0-beta2.apk

hoodFX, 2021-12-20:
in order to solve crop problem, can I use an external sdcard?
Internal storage on my HTC U11 still has around 46GB of free space. is it not enough?
Btw... thanks for 8.0.0 beta 2. i'll try and report if there's any issues.

hoodFX, 2021-12-20:
issues fixed in 8.0.0-beta2. you can close this topic.
now audio work, and crop menu also work. thanks for your hard work.

hoodFX, 2021-12-21:
> Possibly not enough space, I've updated it to hide the manage videos button https://github.com/mirsadm/motioncam/releases/download/8.0.0-beta2/MotionCam.8.0.0-beta2.apk

For audio sync to video, what fps should I use?
I tried recording at 30fps. when it's done, I import it into the davinci resolve video editor with a 30fps timeline too. but the audio duration is shorter than the video duration. the video also runs normally, neither fast nor slow.

closed 2022-01-03 by mirsadm