mirsadm/motioncam #68: Feature request: delete source raw video file after CinemaDNG conversion

nocantsin, 2021-12-20:
It would be really helpful if there was a configuration option that video raw source files (i.e. in Motion Cam's internal format) will get deleted immediately after their CinemaDNG conversion, in order to better deal with storage space limitations. - Alternatively or in addition, it would be great to have an external command line utility for Linux and other operating systems that could do the CinemaDNG conversion on the desktop computer [which would probably be faster, and an alternative way of preventing storage space issues on the phone].

2blackbar, 2021-12-21:
yup , in settings as option

nocantsin, 2021-12-22:
Has been resolved with 8.0.0-beta3 - thank you so much!

closed 2021-12-22 by nocantsin