mirsadm/motioncam #75: [Feature request] compile convert tool too

FilipZamec, 2021-12-21:
Phones get pushed a lot with these huge file sizes. 
One thing how to help is to convert files to dng on PC. 
Would it be possible to give us that convert utility compiled, so we can use it?
I got it in previous versions and it's a big help.

Thank you 🙂

experimentator1709, 2021-12-21:
We are waiting for the utility for converting to Windows PC. converting on a smartphone is a pain!

FilipZamec, 2021-12-21:
Yes, that would be even better, but convert utility already exists and we don't have to wait to dedicated Windows one 🙂

2blackbar, 2021-12-21:
how its called?

FilipZamec, 2021-12-21:
The file is called Convert. You can run it on Linux.

experimentator1709, 2021-12-22:
I was talking about Windows. I and many others are not good at Linux. Also, how does the conversion utility deal with the new VIDEO _ *. Container format?

FilipZamec, 2021-12-22:
I have Windows too. It was recommended to me, that I could use Ubuntu shell from Windows store and it works fine.
New container doesn't work with the old conversion tool I have, that's why I want the actual tool 🙂

experimentator1709, 2021-12-22:
Fine. But your case is not everyone's case. To be popular, an application needs to be friendly. However, the developer promised a conversion utility for Windows.

newillusion, 2021-12-28:
I take participation at this feature request.
I think, a faster way to operatae with .container video is to copy it on a PC and make convertation on the PC.
On my phone (Samsung Note8), converation performs very slow.

mirsadm, 2022-01-04:
I'm going to close this, Windows & Mac tools are available now

closed 2022-01-04 by mirsadm