mirsadm/motioncam #76: Corrections (chromatic aberrations, vignette, geometry distortions) from the RAW metadata were optionally applied when converting to DNG

experimentator1709, 2021-12-21:
It would be nice if corrections (chromatic aberrations, vignette, geometry distortions) from the RAW metadata were optionally applied when converting to DNG. Davinci does not know how to work with this metadata. Correcting distortions in Davinci is not easy for everyone. For example, this is how the vignette fix is added to the DNG at https://sourceforge.net/projects/cornerfix/. Metadata for correction can be turned on and off at https://www.rawdigger.com/.

epicfilmmaker93, 2021-12-22:
Note however that baking in vignetting correction would kill the highlights, and other corrections can only be applied after debayering, so if there is a better debayering algorithm in the future you won't be able to use it

experimentator1709, 2021-12-22:
At least vignette correction like CornerFix does. It will greatly facilitate the work in davinchi. Optionally.

experimentator1709, 2022-01-11:
@mirsadm 
I think you know, but just in case :))
http://www.barrypearson.co.uk/articles/dng/specification.htm#newtags
https://www.adobe.com/content/dam/acom/en/products/photoshop/pdfs/dng_spec_1.4.0.0.pdf