mirsadm/motioncam #84: compilation errors with convert tool

nocantsin, 2021-12-23:
The rewritten convert tool yields the following compiler error messages (and compilation aborts):

`In file included from /home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/source/MotionCam.cpp:1:
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:12:10: error: variable or field ‘ConvertVideoToDNG’ declared void
   12 |     void ConvertVideoToDNG(std::vector<std::unique_ptr<RawContainer> >& containers,
      |          ^~~~~~~~~~~~~~~~~
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:12:33: error: ‘vector’ is not a member of ‘std’
   12 |     void ConvertVideoToDNG(std::vector<std::unique_ptr<RawContainer> >& containers,
      |                                 ^~~~~~
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:8:1: note: ‘std::vector’ is defined in header ‘<vector>’; did you forget to ‘#include <vector>’?
    7 | #include "motioncam/DngProcessorProgress.h"
  +++ |+#include <vector>
    8 |
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:12:45: error: ‘unique_ptr’ is not a member of ‘std’
   12 |     void ConvertVideoToDNG(std::vector<std::unique_ptr<RawContainer> >& containers,
      |                                             ^~~~~~~~~~
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:8:1: note: ‘std::unique_ptr’ is defined in header ‘<memory>’; did you forget to ‘#include <memory>’?
    7 | #include "motioncam/DngProcessorProgress.h"
  +++ |+#include <memory>
    8 |
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:12:68: error: expected primary-expression before ‘>’ token
   12 | d ConvertVideoToDNG(std::vector<std::unique_ptr<RawContainer> >& containers,
      |                                                             ^

/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:12:70: error: expected primary-expression before ‘>’ token
   12 | ConvertVideoToDNG(std::vector<std::unique_ptr<RawContainer> >& containers,
      |                                                             ^

/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:12:73: error: ‘containers’ was not declared in this scope; did you mean ‘RawContainer’?
   12 | vertVideoToDNG(std::vector<std::unique_ptr<RawContainer> >& containers,
      |                                                             ^~~~~~~~~~
      |                                                             RawContaine
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:13:28: error: expected primary-expression before ‘const’
   13 |                            const DngProcessorProgress& progress,
      |                            ^~~~~
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:14:28: error: expected primary-expression before ‘const’
   14 |                            const int numThreads,
      |                            ^~~~~
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:15:28: error: expected primary-expression before ‘const’
   15 |                            const int mergeFrames);
      |                            ^~~~~
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:17:39: error: ‘vector’ in namespace ‘std’ does not name a template type
   17 |     void ConvertVideoToDNG(const std::vector<std::string>& inputFile,
      |                                       ^~~~~~
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:17:34: note: ‘std::vector’ is defined in header ‘<vector>’; did you forget to ‘#include <vector>’?
   17 |     void ConvertVideoToDNG(const std::vector<std::string>& inputFile,
      |                                  ^~~
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:17:45: error: expected ‘,’ or ‘...’ before ‘<’ token
   17 |     void ConvertVideoToDNG(const std::vector<std::string>& inputFile,
      |                                             ^
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:22:10: error: variable or field ‘ConvertVideoToDNG’ declared void
   22 |     void ConvertVideoToDNG(std::vector<int>& fds,
      |          ^~~~~~~~~~~~~~~~~
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:22:33: error: ‘vector’ is not a member of ‘std’
   22 |     void ConvertVideoToDNG(std::vector<int>& fds,
      |                                 ^~~~~~
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:22:33: note: ‘std::vector’ is defined in header ‘<vector>’; did you forget to ‘#include <vector>’?
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:22:40: error: expected primary-expression before ‘int’
   22 |     void ConvertVideoToDNG(std::vector<int>& fds,
      |                                        ^~~
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:23:28: error: expected primary-expression before ‘const’
   23 |                            const DngProcessorProgress& progress,
      |                            ^~~~~
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:24:28: error: expected primary-expression before ‘const’
   24 |                            const int numThreads=4,
      |                            ^~~~~
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:25:28: error: expected primary-expression before ‘const’
   25 |                            const int mergeFrames=0);
      |                            ^~~~~
In file included from /home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/source/MotionCam.cpp:1:
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:30:33: error: ‘vector’ in namespace ‘std’ does not name a template type
   30 |     void GetMetadata(const std::vector<int>& fds, float& outFrameRate, int& outNumFrames, int& outNumSegments);
      |                                 ^~~~~~
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:30:28: note: ‘std::vector’ is defined in header ‘<vector>’; did you forget to ‘#include <vector>’?
   30 |     void GetMetadata(const std::vector<int>& fds, float& outFrameRate, int& outNumFrames, int& outNumSegments);
      |                            ^~~
/home/paragram/Downloads/software/motioncam/libMotionCam/libMotionCam/include/motioncam/MotionCam.h:30:39: error: expected ‘,’ or ‘...’ before ‘<’ token
   30 |     void GetMetadata(const std::vector<int>& fds, float& outFrameRate, int& outNumFrames, int& outNumSegments);
      |                                       ^
make[2]: *** [libmotioncam/CMakeFiles/motioncam-static.dir/build.make:1210: libmotioncam/CMakeFiles/motioncam-static.dir/libMotionCam/source/MotionCam.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:126: libmotioncam/CMakeFiles/motioncam-static.dir/all] Error 2
make: *** [Makefile:91: all] Error 2`

mirsadm, 2021-12-23:
Cheers will look at this. Will try compiling in Ubuntu instead of macos

FilipZamec, 2021-12-25:
And once it works, make it available to all of us 🙂
Thanks