mirsadm/motioncam #85: Orientation sensor

willybanana, 2021-12-24:
Orientation sensor is too sensitive, it would be convenient to have a forced setting.
+ fix overlaps

[image]
[image]

newillusion, 2021-12-28:
I have the issue with horisontal orientation. Samsung Note8.
P.S. Previous buttons (from first beta version) was better solution. 
[image]

mirsadm, 2021-12-28:
Yes that does need to be fixed. I am intending on putting buttons back but making them configurable. As I add more features they're not going to fit there but ultimately I imagine only a few things are useful without having to access the extended menu.

2blackbar, 2021-12-28:
I think now its cleaner, You can hide it with tap on that arrrow on left.
It would be cool to have a global setting to hide all these menus with shutter and iso etc when recording tho.
Not by default of course.Just a switch.

newillusion, 2021-12-28:
> I am intending on putting buttons back but making them configurable.
Sounds great! It helps to time and attention economy (time is very valueable resource during video shooting). Thanks!
And if user can choose what exactly place on "hot" place - that is great twice...

teemodk, 2022-01-07:
Not a big issue for photos, but it should not move during recording of videos. Especially for Zenfone 6/7/8; when lens is turned 90° (useful when recording Antz), the display is horizontal and it flickers very fast back and forth between orientation modes.

mirsadm, 2022-01-07:
Yep I will to add an orientation lock, make is less sensitive and rotate to UI after I release the current build

mirsadm, 2022-01-08:
Fixed the UI issues with the settings not being oriented correctly:
https://github.com/mirsadm/motioncam/releases/download/8.0.7-beta0/MotionCam.8.0.7-beta0.apk

I will make the orientation changes less sensitive/prevent oscillation in the next release