mirsadm/motioncam #88: out of focus photos or videos

nerdpaolino, 2021-12-25:
I can't focus properly. Every time i get blurry/soft images. If I shoot with another app like opencamera the same subject with same settings everything is ok. This problem is present in photo and video mode. Redmi note 10 pro.

nerdpaolino, 2021-12-26:
I'm closing the issue. I've found that the image need to be sharpened with the "sharpen" tool in Darktable.
With Radius=2 Amount=2 Threshold=0 I get an image that is as sharp as the JPG created with the DNG.
It wasn't a focus problemafter all, but in my opinion the focus tool is very difficult to use because there isn't an option to zoom a screen area.
Thanks for the hard work.

closed 2021-12-26 by nerdpaolino

mirsadm, 2021-12-26:
You were correct, the auto focus was stuck to infinity until toggled on/off. I've fixed it. Publishing to play store + uploading to GitHub now.