mirsadm/motioncam #91: The camera can't see anything since 8.0.0-beta0

mido830, 2021-12-26:
I use Sony Xperia 5 II.
This is how the app looked like since 7.0 to 7.0.8-beta5:
[image]

This is how it looks like since 8.0.0-beta0:
[image]

It's just not that the preview doesn't work, the camera can't capture anything at all, be it photos or videos.

mirsadm, 2021-12-26:
Could you try selecting RAW16 or RAW10 in the settings to see if it changes anything?

mido830, 2021-12-26:
> Could you try selecting RAW16 or RAW10 in the settings to see if it changes anything?

Just tested, no difference.

mirsadm, 2021-12-26:
Could you give this one a go please:
https://github.com/mirsadm/motioncam/releases/download/8.0.3/MotionCam.8.0.3.apk

mido830, 2021-12-26:
> Could you give this one a go please: https://github.com/mirsadm/motioncam/releases/download/8.0.3/MotionCam.8.0.3.apk

I tested all releases, this one included. Not a single release work for me since 8.0.0-beta0

mirsadm, 2021-12-26:
hmm I'd need some logs from ADB to figure out what's going on. Have you also tried uninstalling and installing it again so it resets all the settings?

mido830, 2021-12-26:
> hmm I'd need some logs from ADB to figure out what's going on.

How do I generate them? Max I was using ADB for was removing bloatware.

mirsadm, 2021-12-26:
https://forum.xda-developers.com/t/tutorial-how-to-logcat.1726238/

Basically this line:
>adb logcat -v long > logs.txt

Run then run the app maybe switch cameras and exit. The logs should be in logs.txt

mido830, 2021-12-26:
[logs.txt](https://github.com/mirsadm/motioncam/files/7776959/logs.txt)

mirsadm, 2021-12-26:
Thanks for the logs. I think I've narrowed down the change since then. Here's another release:
https://github.com/mirsadm/motioncam/releases/download/8.0.4/MotionCam.8.0.4.apk

mido830, 2021-12-26:
Fixed in 8.0.4.

closed 2021-12-26 by mido830