mirsadm/motioncam #92: Need the lowest quality JPG preview for the DNG

experimentator1709, 2021-12-26:
So that you can quickly view what is contained in large file folders. 
Optional.

2blackbar, 2021-12-26:
Thats very vague, dng photo or all raw video frames? you can do this on pc with batch convert in xnview to jpeg.
kinda like proxy

experimentator1709, 2021-12-26:
Primarily in the video. 

But if not in the photo, then it will not hurt there either.

2blackbar, 2021-12-26:
Yeah that would make converting take longer, you can do this easily on pc, with a few clicks in xnview batch processing for folder with dngs,.

experimentator1709, 2021-12-27:
Thanks for the tip. But my scenario is as follows - while traveling, I record a rav on an external SSD with a size of 1 terabyte (this is about 60 minutes of 4K), then at the hotel I use a smartphone to convert it to DNG. There is no computer. Time is all night. The next morning I look at the resulting DNG on my smartphone and delete the unnecessary or not obtained. I will clarify again - this is optional and should be enabled in the settings as desired.

2blackbar, 2021-12-27:
I just converted dngs to jpeg with batch image converter from playstore