mirsadm/motioncam #93: Detection area extends too much under the shutter button

Thundermaxx, 2021-12-26:
Thank you so much for the clipping warning overlay in photo mode. Really easy to get good exposure and the red color is perfectly visible. Sharpness setting is a nice add-on too. Could it be possible to add a setting to automatically reset dual exposure controls between each capture?

Could you please restrict the detection area of the shutter button to the white circle as it extends under it. This interferes with the swipe gesture on my s20u. When I swipe up to exit the app, it takes a picture.

Since 8.03, autofocus doesn't work immediately after install. You need to lock and unlock the AF first to make it work. This problem maybe specific to s20

Do raw10/12/16 settings apply to photo mode?

I know the DNG video mode is the big game right now but the photo mode is so good and I really appreciate the efforts you put into.

mirsadm, 2022-01-08:
> Do raw10/12/16 settings apply to photo mode?

These apply to all modes

This should fix the rest of your issues:
https://github.com/mirsadm/motioncam/releases/download/8.0.7-beta0/MotionCam.8.0.7-beta0.apk

closed 2022-01-08 by mirsadm

Thundermaxx, 2022-01-10:
Hi !

Thx for all.

Stephan Buchin

https://github.com/mirsadm/motioncam/releases/download/8.0.7-beta0/MotionCam.8.0.7-beta0.apk