mirsadm/motioncam #94: [Feature request] Different color for different level of highlights clipping

amitkattal, 2021-12-27:
We know with RAW, using highlight recovery, to some level clipped highlights could be recovered. It would be nice to have a different color for different levels of clipping like dark red for very clipped but like green for areas clipped but still recoverable (Maybe like +1ev) .

epicfilmmaker93, 2021-12-30:
> to some level clipped highlights could be recovered

Only if, for example, green and blue pixels clipped and red did not; successful recovery depends on the program you are using. It's either clipped or not clipped, and I think that's how it works currently.

mirsadm, 2021-12-30:
At the moment I'm using the highest channel value from the RAW pixels and drawing it based on a simple curve. This can be tweaked to be more gradual