mirsadm/motioncam #96: 2 questions on DNG. Compression and White Level. (RAW video)

experimentator1709, 2021-12-27:
---- ExifTool ----
ExifTool Version Number      : 12.25
---- File ----
File Name                    : frame-000074.dng
Directory                    : C:/Users/MASTER/Desktop
File Size                    : 6.1 MiB
File Type                    : DNG
MIME Type                    : image/x-adobe-dng
Exif Byte Order              : Little-endian (Intel, II)
---- EXIF ----
Subfile Type                 : Full-resolution image
Image Width                  : 4000
Image Height                 : 3000
Bits Per Sample              : 16

(1)
Compression                  : JPEG

Using lossy compression? What do I lose in RAW? 

(2)
White Level                  : 1023

Does it indicate a 10 bit signal from the matrix?

2blackbar, 2021-12-27:
Youd have to analyze actual raw frames for that, there available in builds pre alpha 7.
I agree about the compression tho, would be great to have optional uncompressed dng frames when exporting but for me  i prefere dngs with compression.Dynamic range of dngs is lost when encapsulating frames into rec.709 signal. and that doesnt occur in motioncam.

mirsadm, 2021-12-27:
The white level is whatever the camera reports. In your case it's 10 bit. 

It should not be lossy, that might be referring to the JPEG preview image which is saved along the RAW data. I'm using the Adobe DNG SDK to write the DNGs, it seems to apply compression to the DNG regardless of the option I pass to it, it is almost certainly lossless though.

closed 2021-12-29 by mirsadm