mirsadm/motioncam #97: 0.container format question

jdorman12, 2021-12-27:
Hello mirsadm, greatly appreciate the work on this app, it is honestly a work of art..

My question is from the original raw video io stream/container that is a custom format, can we convert that straight to video without the conversion to dng? Tried importing the dngs to resolve but the sequence was giving me problems but I didn't mess with it too long yet. Also read that there was a converter on Linux maybe and was wondering where we could get that, thank you very much and i really enjoy this app, thank you for your hard work.

mirsadm, 2021-12-27:
Hi,
Thanks for the kind words. Unfortunately there are very few options for open file format standards when it comes to RAW video. I realise the most annoying part is the conversation at the moment. My current plan is to write some GUI tools for Windows and Mac which would let you convert the video on a PC. I'm working on that now. I'll charge a small for fee for those which will let me keep the app free and help with development. There is a command line tool that is part of the source that you can compile and use to convert the container as well. That's going to be free but I don't have the will not energy to compile and distribute that in way that'll work for everybody, especially windows which is quite annoying to work with.

nocantsin, 2021-12-28:
@jdorman12  The (commercial) program FastDNG Processor does exactly what you are looking for:  https://www.fastcinemadng.com 

Alternatively, you can use the gratis version of Davinci Resolve to import CinemaDNG and export it as conventional video.

nocantsin, 2021-12-28:
@mirsadm : Another solution could be to offer transcoding MotionCam's internal raw video format to MagicLantern's MLV, next to CinemaDNG and use the very capable Open Source program MLV App for color correcting and exporting conventional video. Or, alternatively, add CinemaDNG support to MLV App.

https://mlv.app

FilipZamec, 2021-12-28:
I like the paid app for converting the raw to dngs. Happy to pay for your work. It's revolutionary.

mirsadm, 2021-12-29:
@nocantsin mlv is an interesting one and would be feasible. 

I think once I release some tools for Windows this pain point will largely be gone. One benefit of using this custom format, besides performance, is that it doesn't restrict me for the future. One thing I want to do is provide a near lossless compression option which might open up the possibility of much higher frame rates and resolutions. Getting that to work would not really be feasible otherwise.

FilipZamec, 2021-12-29:
Looking forward to higher framerates 👌

pauloup, 2021-12-29:
> One thing I want to do is provide a near lossless compression option which might open up the possibility of much higher frame rates and resolutions.
@mirsadm That's super exciting!

jdorman12, 2021-12-29:
If anyone could upload a video or step by step instructions for a dummy for how to do this in windows I would appreciate it, I don't have a clue how to convert or run this..

mirsadm, 2021-12-30:
It's not ready just yet. I'll put it up the next few days I hope. It should be fairly straightforward to use.

2blackbar, 2021-12-30:
exporting to mlv would be great!

mirsadm, 2021-12-30:
If anyone is interested I have made a very very early rough windows application to convert the files to DNG here:
https://github.com/mirsadm/motioncam/releases/download/8.0.4/motioncam-tools-windows-0.0.2.zip

nocantsin, 2021-12-31:
It runs under Linux with Wine, but crashes at 81% of the conversion (and leaves an incomplete yet working DNG folder).

Here's Wine's error message:

Unhandled exception: page fault on read access to 0x000000000000000c in 64-bit code (0x00000001400954a0).
Register dump:
 rip:00000001400954a0 rsp:000000000536f580 rbp:000000000536f680 eflags:00010202 (  R- --  I   - - - )
 rax:0000000000000000 rbx:000000000c3c9c80 rcx:000000000bfefd30 rdx:0000000000000001
 rsi:0000000000000000 rdi:000000000524f8b0  r8:0000000000000000  r9:0000000000000002 r10:00000001400fd142
 r11:000000000c416db8 r12:0000000000000c00 r13:000000000c2f03e8 r14:0000000000000004 r15:0000000000000ff0
Stack dump:
0x000000000536f580:  0000000200000002 000000000c3c9c80
0x000000000536f590:  000000000524f8b0 0000000000000000
0x000000000536f5a0:  00000000000000ff 0000000000000001
0x000000000536f5b0:  0000000000000000 0000000000000000
0x000000000536f5c0:  000000000536f6e0 000000017002b12b
0x000000000536f5d0:  0000002100000019 0000000040000000
0x000000000536f5e0:  0000000000000001 0000000000000000
0x000000000536f5f0:  0000000000000000 000000017004e6a9
0x000000000536f600:  0000000000000bff 0000000003a5ee10
0x000000000536f610:  0000000200000000 00000ff000010000
0x000000000536f620:  000000000c3c9c80 000000000524f8b0
0x000000000536f630:  000000000536fc20 000000000536fbb0
Backtrace:
=>0 0x00000001400954a0 in motioncam-tools (+0x954a0) (0x000000000536f680)
  1 0x0000000140095ce5 in motioncam-tools (+0x95ce5) (0x0000000000000000)
  2 0x000000014008c036 in motioncam-tools (+0x8c036) (0x0000000000000000)
  3 0x0000000140087a6b in motioncam-tools (+0x87a6b) (0x0000000000000000)
  4 0x00000003af6dc53b _beginthreadex_trampoline+0x4b(arg=0000000003A59DC0) [Z:\usr\src\packages\BUILD\dlls\msvcrt\thread.c:199] in ucrtbase (0x0000000000000000)
  5 0x000000007b62c709 BaseThreadInitThunk+0x9(unknown=<internal error>, entry=<internal error>, arg=<internal error>) [Z:\usr\src\packages\BUILD\dlls\kernel32\thread.c:61] in kernel32 (0x0000000000000000)
  6 0x000000017005b633 __wine_pop_frame(unknown=<internal error>, entry=<internal error>, arg=<internal error>) [Z:\usr\src\packages\BUILD\include\wine\exception.h:273] in ntdll (0x0000000000000000)
  7 0x000000017005b633 RtlUserThreadStart+0x83(entry=[<register RSP not accessible in this frame>, arg=[<register RSP not accessible in this frame>) [Z:\usr\src\packages\BUILD\dlls\ntdll\thread.c:240] in ntdll (0x0000000000000000)
0x00000001400954a0 motioncam-tools+0x954a0: movd	0x000000000000000c(%rax),%mm0
Modules:

mirsadm, 2021-12-31:
Ah fixed that problem. Haven't programmed for Windows in a long time, it's full of "quirks".

mirsadm, 2021-12-31:
That's been fixed here:
https://github.com/mirsadm/motioncam/releases/download/8.0.4/motioncam-tools-0.0.3-windows.zip

nocantsin, 2021-12-31:
This works great, even under Wine in Linux. Thank you! The only small issue is that it doesn't also  move the .wav files into the corresponding DNG folders.

mirsadm, 2021-12-31:
Yes it should definitely match what the app does. Happy new year!

experimentator1709, 2021-12-31:
@mirsadm 
Happy new year!

nocantsin, 2021-12-31:
Happy new year! IMHO, MotionCam was the greatest Open Source software filmmaking gift to the world in 2021!

mirsadm, 2022-01-03:
I'm going to close this one for now
thanks for the kind words @nocantsin

closed 2022-01-03 by mirsadm