mirsadm/motioncam #98: framerate & audio synchronization issues (Raw Video)

nocantsin, 2021-12-28:
Not sure which of the following issues are specific to my phone hardware (Google Pixel 5) and which ones might be fixable in Motion Cam's code:

- Not all framerates offered in the Raw Video module do actually work. So far, I've had success with 25p and 30p; in those settings, the camera records the specified framerate with synchronous audio. At 24p, however, the camera actually records 30p and tags it as 24p so that it plays back as 80% slo-mo with asynchronous sound. [If that footage is conformed back to 30p, video and sound will be synchronous.] I wouldn't be surprised, though, if those framerate limitations were a limitation of my phone hardware (since 25p and 30p are the standard PAL/European and NTSC/American tv and video framerates). 
- Even when choosing fully working framerates, audio recording starts several frames (ca. 6.5 frames at 25p) later than CinemaDNG video recording. As a result, Davinci Resolve doesn't recognize the Motion Cam's raw video recordings as proper CinemaDNG [i.e. raw video with sychronous audio], but imports the DNG image sequence and the .wav audio file as two separate files/assets. In the timeline, one then needs to manually align each audio track with the end of its corresponding video clip to have video with sound. Maybe this could be avoided by slightly delaying the beginning of a new video recording so that audio can catch up with video? 
- There can be framedrops in raw video/CinemaDNG material recorded with Motion Cam, which become visible as jumps during video playback. This also happens when recording in lower resolutions with a cropped sensor. It would be very nice if Motion Cam [in case it has ways of detecting framedrops during recording at all] had a configurable option in its setup menu to automatically stop the recording of a raw video clip whenever a frame has been dropped. 

That said, Motion Cam is truly great if not groundbreaking. I am extremely grateful for all your fantastic work!

mirsadm, 2021-12-29:
-Not all frame rates offered in the Raw Video module do actually work

This is unfortunately the reality of working with Android. There is no consistently in behaviour. I think at best I could detect that the selected frame rate is not being delivered

-Even when choosing fully working frame rates, audio recording starts several frames (ca. 6.5 frames at 25p) 

Yes this needs to be synchronised. I didn't realize that Resolve treated them as a pair. I can ensure they start at the same time.

-There can be framedrops in raw video/CinemaDNG material recorded with Motion Cam

In the earlier versions I did have a frame drop counter. I will need bring this back. 

Thanks for the feedback, it is very helpful!

2blackbar, 2021-12-30:
I dont think its that, my phone has 24fps available in all other apps that support it like filmic pro, opencamera , cinema4k etc...
But motioncam's way of acquiring  lets say 24fps is by slowing down the footage shot in 30fps, it still records 30 frames per second and probably chops off the end to get as many frames as with 24.
Why? Cause when i synchronise the voice at the beginning of the video, then it gets desynchronised after second or two, sound plays faster, that means footage  has too many frames and is in slight slowmotion cause of it.
Theres no actual 24fps in motioncam, theres 30fps  and maybe it needs different approach to acquire it.

mirsadm, 2021-12-30:
I'll look into this further. There's probably some better way to get the camera to deliver the frame rate with correct frame pacing

teemodk, 2021-12-30:
Just ran 10 seconds of 5,12,24,25 and 30fps on Zenfone 6. All with correct number of frames recorded.

2blackbar, 2022-01-02:
OK i think its something with manual exposure, there might be some error in the code, because when i leave autoexposure, i can record proper 24fps and sound is synced on my phone, but when i press shutter to lock it to 50, then it gets desynchronised with audio when merging.
So there clearly is something about the code where its switching from autoexposure to manual exposure, on manual it stays in 30fps, not on all phones but on some.
The code for 24fps ,12fps is good on auto so maybe using it as starting point would solve it.

mirsadm, 2022-01-02:
I've put up another beta release with new frame rate options. Previously the app didn't query the camera for the frame rates that it supports. Now it does that. 
@nocantsin Hopefully this should fix your problem because it should only list the frame rates the device actually supports.
@2blackbar I suspect choosing a manual exposure outside of the listed ranges will result in this sort of behaviour. I need to investigate this further.

nocantsin, 2022-01-02:
@mirsadm Thanks! - I am getting a bug though: The "Manage Videos" dialogue crashes directly after having generated the thumbnails for the recorded clips.

mirsadm, 2022-01-02:
Oh dear, could you get the ADB logs please?

nocantsin, 2022-01-02:
Here it is:

[8.0.6-beta_crash_log.txt.gz](https://github.com/mirsadm/motioncam/files/7799971/8.0.6-beta_crash_log.txt.gz)
!

mirsadm, 2022-01-02:
Thanks, was that the only video available (recorded from the new version of the app)?

nocantsin, 2022-01-02:
Yes, I emptied the directory before to go sure (first thought that the app would crash on files with the old compression format). Will try once more and see whether I can reproduce the behavior.

mirsadm, 2022-01-02:
Cheers. What was your output resolution, was there any cropping? Just trying to narrow it down.

nocantsin, 2022-01-03:
Output resolution was 2732x1842 with 33% crop width and 40% crop height.

I can fully reproduce the behavior (with an emptied directory), will now try again without sensor crop.

nocantsin, 2022-01-03:
Getting exactly the same behavior with an uncropped sensor. Here's a new log - taken while recording to brief raw videos and repeatedly going in to the "Manage Videos" dialogue afterwards:

[8.0.6-beta_crash_log-2.txt.gz](https://github.com/mirsadm/motioncam/files/7799984/8.0.6-beta_crash_log-2.txt.gz)

mirsadm, 2022-01-03:
Thanks, I'll look at that tomorrow. By the way does the frame rate selection match up to the frame rates that you've found to work correctly before?

nocantsin, 2022-01-03:
On the PIxel 6 Pro, it only shows me 15, 30 and 60 as working frame rates, although 24p and 25p work on that phone too (24p didn't work on the PIxel 5)

mirsadm, 2022-01-03:
@nocantsin Have finally gotten to the bottom of the frame rate audio/video sync problems. Should be fixed in the next update.

mirsadm, 2022-01-03:
That's up now
https://github.com/mirsadm/motioncam/releases/download/8.0.6-beta2/MotionCam.8.0.6-beta2.apk

Will wait to confirm it is fixed before closing this

nocantsin, 2022-01-03:
Unfortunately, now Raw Video doesn't work at all any more. When switching to the "Raw Video" tab, I get this error popup message:

[image]

I will create new log in a few minutes.

mirsadm, 2022-01-03:
Wow that's unexpected. Could you try clearing the apps storage/cache?

nocantsin, 2022-01-03:
I'
[motion_cam-error-log.txt.gz](https://github.com/mirsadm/motioncam/files/7802423/motion_cam-error-log.txt.gz)
m actually getting the error in all tabs/modules of the app, also in "Photo" (after a few seconds of having started the app). Here is my log file. 

With app storage/cache, do you mean the directory where the raw video is being stored, or other directories as well?

mirsadm, 2022-01-03:
You could uninstall/reinstall the app which would have the same result. Otherwise I would also try restarting the phone. I'm struggling to think of any changes that would result in this happening

nocantsin, 2022-01-03:
Same behavior/error message/crash after deinstalling the app, restarting the phone and reinstalling 8.0.6 beta 2.

Here is a simple log, made from nothing but starting up Motion Cam and waiting for the error message to appear and the app to crash (without switching to the Raw Video tab):

[motion_cam-8.0.6beta2-simple_crash_error-log.txt.gz](https://github.com/mirsadm/motioncam/files/7802469/motion_cam-8.0.6beta2-simple_crash_error-log.txt.gz)

mirsadm, 2022-01-03:
Thanks for the logs. The old version still works? I'm a bit dumbfounded

mirsadm, 2022-01-03:
Ok I think I know what is causing that issue. I suspect it's a bug in the camera service on your phone. 
https://github.com/mirsadm/motioncam/releases/download/8.0.6-beta2/MotionCam.8.0.6-beta3.apk

nocantsin, 2022-01-03:
The camera now works again, but the app crashes as soon as I access the "Manage Videos" file and it has finished generated the video thumbmails.  I deinstalled the app, cleared the storage folder, reinstalled, but got the same behavior. Log file:

[motion_cam-8.0.6beta3-simple_crash_error-log.txt.gz](https://github.com/mirsadm/motioncam/files/7802597/motion_cam-8.0.6beta3-simple_crash_error-log.txt.gz)

mirsadm, 2022-01-03:
I've updated the windows tools so they should be able to convert the files in the meantime

nocantsin, 2022-01-03:
Here's a sample container file:
[http://data.pleintekst.nl/VIDEO_20220103_161735.0.container](http://data.pleintekst.nl/VIDEO_20220103_161735.0.container)

- the Windows conversion tool indeed helps as a workaround.

mirsadm, 2022-01-03:
Thanks for that. I ended up finding the issue. I assume this was the Pixel that was crashing? Really every other phone should have as well, makes me think that Google is the only manufacturer taking security seriously. Anyway here it is:
https://github.com/mirsadm/motioncam/releases/download/8.0.6-beta4/MotionCam.8.0.6-beta4.apk

In addition you can disable/configure the compression in settings as well

nocantsin, 2022-01-03:
Yes, this update fixed the issue of the CinemaDNG conversion dialogue crashing. So many thanks for all your hard work!

mirsadm, 2022-01-03:
👍 great to hear. I shall close this issue for now. Let me know if you still run into audio/video sync issues.

closed 2022-01-03 by mirsadm