mirsadm/motioncam #100: only main camera works

nocantsin, 2021-12-29:
Probably a known issue: in all operation modes, Motion Cam does recognize the available camera modules and can switch between front and rear; however, on the rear, only the main camera is supported. When switching to other available rear camera modules (such as tele and wide-angle), the display remains dark.

2blackbar, 2021-12-30:
what phone? on oneplus 8t it works with all cams, its related to phone models.
It is a nightmare to support all phones out there and takes time, especially if dev doesnt have that phone.

nocantsin, 2021-12-30:
In my case, I tried both a Google Pixel 5 and Pixel 6 Pro.

mirsadm, 2021-12-30:
Hmm I'm surprised it's the pixel as they're usually quite good with that sort of stuff. Are you able to get logs for me with ADB?

nocantsin, 2021-12-30:
Just took a crash course in ADB and have it running and connected with my machine. But what do I need to do to get logs from MotionCam via ADB?

mirsadm, 2021-12-30:
This guide is helpful
https://forum.xda-developers.com/t/tutorial-how-to-logcat.1726238/

This line in particular
adb logcat -v long > problem.txt

Then if you the app and switch cameras it should log it

nocantsin, 2021-12-30:
Thanks, this solved it!

I created two log files, one for switching from the main camera to the tele camera (log1.txt) and one for switching from the main camera to the wide angle camera (log2.txt)

[motioncam_log.tar.gz](https://github.com/mirsadm/motioncam/files/7793840/motioncam_log.tar.gz)

mirsadm, 2021-12-30:
Thanks for doing that! will take a look soon when I get a chance.

2blackbar, 2021-12-30:
My cameras are crashing motioncam only if i give too much ram to the app, so with 4 gb, maximum is 1800, more and it can shutdown app randomly

ScherbakovAl, 2022-01-03:
xiaomi mi 8 lite won't switch to front camera. Focusing does not work on the main camera.

mirsadm, 2022-01-05:
@nocantsin I forgot to ask. 
a) if you select burst mode and capture a photo (even if the screen is black) does it show any captured images?
b) If you enable dual exposure in settings, do you still get a black screen for the other camera

nocantsin, 2022-01-05:
@mirsadm 
a) burst mode doesn't capture any images with the other (no-supported cameras). The screen is black, but even if I shoot anyway, no image results.
b) enabling dual exposure crashes the app.  So badly, that the app will crash immedaitely after every new launch, and the only way to remedy the situation is to deinstall and reinstall the app to get rid of this setting. 

I had behavior b) with previous versions of MotionCam, but at some point simply never touched this option again. (And ignored it, because I had thought that it was unrelated to raw video recording.)

Here's my log file of the app crashing with the activated dual exposure setting:

[motion_cam-error-crash_log.txt.gz](https://github.com/mirsadm/motioncam/files/7818084/motion_cam-error-crash_log.txt.gz)

LuckyBastard73, 2022-01-13:
I have the same issue on the Xiaomi Mi Note 10 as well.
Only the main camera is shown. 
In GCam all modules on the back can be accessed.
Also ProShot can see all cameras. 
Most other camera apps can also see only the main camera.

I would like to work with the other camera modules as they have a lower resolution than the man sensor.
The Mi Note 10 can not handle the resulting 27MP from the main 108MP sensor. 
When I activate pixel binning the resulting DNGs are not correct, so this is also not an option.

There is a little tool to read out the camera IDs and this is the result:

============================

Device : Xiaomi Mi Note 10 (tucana)
Manufacturer : Xiaomi
Android : 11
Fingerprint : Xiaomi/tucana_eea/tucana:11/RKQ1.200826.002/V12.5.5.0.RFDEUXM:user/release-keys

============================

Camera IDs Visible to Apps = [0, 1]

============================

All Camera IDs = [0, 1, 20, 21, 22, 23, 60, 61, 62, 63, 100, 101, 102, 120, 180]

============================

CameraID = [0]  ★
Facing = BACK
Zoom = 1x
Type = MAIN
FocalLength = 6.72mm
35mm eqv FocalLength = 25.13mm
Aperture = 1.69
SensorSize = 9.6256x7.2192
PixelArray = 6016x4512
PixelSize = 1.60µm
AngleOfView(Diagonal) = 110°
AEModes = [0, 1, 2, 3, 4]
FlashSupported = true
RAW_SENSOR sizes = [6016x4512]
SupportedHardwareLevel = 3

============================

CameraID = [1]  ★
Facing = FRONT
Zoom = 1x
Type = MAIN
FocalLength = 3.85mm
35mm eqv FocalLength = 26.41mm
Aperture = 2.0
SensorSize = 5.248x3.9424
PixelArray = 3280x2464
PixelSize = 1.60µm
AngleOfView(Diagonal) = 108°
AEModes = [0, 1]
FlashSupported = false
RAW_SENSOR sizes = [3280x2464]
SupportedHardwareLevel = 3

============================

CameraID = [20]
Facing = BACK
Type = LOGICAL
FocalLength = 8.16mm
35mm eqv FocalLength = 52.04mm
[... 193 lines cut ...]
PixelSize = 1.60µm
AngleOfView(Diagonal) = 110°
AEModes = [0, 1, 2, 3, 4]
FlashSupported = true
RAW_SENSOR sizes = [6016x4512, 5184x3880, 4032x3024]
SupportedHardwareLevel = 3

============================

CameraID = [180]
Facing = BACK
Type = LOGICAL
FocalLength = 6.72mm
35mm eqv FocalLength = 25.13mm
Aperture = 1.69
SensorSize = 9.6256x7.2192
PixelArray = 6016x4512
PixelSize = 1.60µm
AngleOfView(Diagonal) = 110°
AEModes = [0, 1, 2, 3, 4]
FlashSupported = true
RAW_SENSOR sizes = [6016x4512, 5184x3880, 4032x3024, 3264x2448]
SupportedHardwareLevel = 3

============================

mirsadm, 2022-01-13:
I have a feeling it's because the manufacturer has purposely disabled them so they are not discoverable. They may work if I add an option to manually define the camera ids to try opening in the app. Worth a test

LuckyBastard73, 2022-01-13:
They seem hidden. But it is the same situation as for the ZTE Axon 30 Ultra where only the main camera is visible to typical camera apps (in this case also for every GCam if tried). Luckyly all modules are detected for the Axon 30 Ultra.

============================
Camera IDs Visible to Apps = [0, 1]
============================
All Camera IDs = [0, 1, 2, 3, 4, 5]
============================
CameraID = [0]  ★
Facing = BACK
Zoom = 1x
Type = MAIN
FocalLength = 5.58mm
35mm eqv FocalLength = 27.15mm
Aperture = 1.6
SensorSize = 7.3984x5.5552
PixelArray = 4624x3472
PixelSize = 1.60µm
AngleOfView(Diagonal) = 106°
AEModes = [0, 1, 2, 3]
FlashSupported = true
RAW_SENSOR sizes = [4624x3472]
SupportedHardwareLevel = 3
============================
CameraID = [1]  ★
Facing = FRONT
Zoom = 1x
Type = MAIN
FocalLength = 3.74mm
35mm eqv FocalLength = 28.89mm
Aperture = 2.45
SensorSize = 4.660992x3.499776
PixelArray = 2312x1736
PixelSize = 2.02µm
AngleOfView(Diagonal) = 115°
AEModes = [0, 1]
FlashSupported = false
RAW_SENSOR sizes = [2304x1728]
SupportedHardwareLevel = 3
============================
CameraID = [2]  ★
Facing = BACK
Zoom = 0.53x
Type = ULTRAWIDE
FocalLength = 2.59mm
35mm eqv FocalLength = 14.35mm
Aperture = 2.2
SensorSize = 6.496x4.8608
PixelArray = 4640x3472
PixelSize = 1.40µm
AngleOfView(Diagonal) = 131°
AEModes = [0, 1, 2, 3]
FlashSupported = true
RAW_SENSOR sizes = [4640x3472]
SupportedHardwareLevel = 3
============================
CameraID = [3]  ★
Facing = BACK
Zoom = 1.20x
Type = TELE
FocalLength = 5.88mm
35mm eqv FocalLength = 32.59mm
Aperture = 1.9
SensorSize = 6.496x4.8608
PixelArray = 4640x3472
PixelSize = 1.40µm
AngleOfView(Diagonal) = 88°
AEModes = [0, 1, 2, 3]
FlashSupported = true
RAW_SENSOR sizes = [4640x3472]
SupportedHardwareLevel = 3
============================
CameraID = [4]  ★
Facing = BACK
Zoom = 4.67x
Type = TELE
FocalLength = 11.60mm
35mm eqv FocalLength = 126.93mm
Aperture = 3.4
SensorSize = 3.290112x2.467584
PixelArray = 3264x2448
PixelSize = 1.01µm
AngleOfView(Diagonal) = 20°
AEModes = [0, 1, 2, 3]
FlashSupported = true
RAW_SENSOR sizes = [3264x2448]
SupportedHardwareLevel = 3
============================
CameraID = [5] = [0+2+3+4]
Facing = BACK
Type = LOGICAL
FocalLength = 5.58mm
35mm eqv FocalLength = 27.15mm
Aperture = 1.6
SensorSize = 7.3984x5.5552
PixelArray = 4624x3472
PixelSize = 1.60µm
AngleOfView(Diagonal) = 106°
AEModes = [0, 1, 2, 3]
FlashSupported = true
RAW_SENSOR sizes = [4624x3472]
SupportedHardwareLevel = 3
============================

nocantsin, 2022-01-13:
In the case of the Pixel 5 and 6 Pro, Motion Cam seems to detect the other cameras, but when switching to them, the screen remains black (until one switches back to the main camera).

week5thor, 2022-01-21:
LG V20 H910, android 8.0.0
Wide camera don't working

lbdroid, 2022-01-26:
In the case of Pixel 6 Pro, the HAL is set up to "seamlessly" switch between camera sensors based, in part, on the zoom level selected.

The wide angle sensor is selected for all zoom levels LESS than 1:1 (for example, 0.8:1).
The main sensor is selected from 1:1 through about 4.3:1.
The telephoto sensor is selected from 4.3:1, IF the subject is far enough for the camera to focus on.

I'm not sure if the HAL actually allows you to select the camera individually, which means that it will definitely do digital scaling if you don't catch the exact level of zoom.