mirsadm/motioncam #99: Using Motion Cam on an Asus ZenFone 7 Pro smartphone

lordsvarogofmax, 2021-12-29:
Hi! My name is Max, I'm from Russia. I am using Motion Cam on an Asus ZenFone 7 Pro smartphone and faced with one situation.

My phone is equipped with 3 cameras - a body camera, a main camera and a wide-angle camera. 
The cameras are located on the rotary unit, which means that when switching the front camera, the unit simply turns and uses one of the cameras as the front one.

Characteristics of the Asus ZenFone 7 Pro camera module:

Primary: 64 MP 1/1.72-inch Quad-Bayer sensor, 26 mm-equivalent lens with f/1.8 aperture, PDAF and OIS
Telephoto: 8MP 80 mm-equivalent (3x optical) lens with f/2.4 aperture, PDAF and OIS
Ultra-wide: 12 MP 1/2.55-inch sensor, 16.6 mm-equivalent lens with f/2.2 aperture, dual pixel PDAF

Now to my situation.

When using the Primary camera, the maximum video resolution is 4,576 x 3,432. Without binning.
When the front mode is activated, the camera selection disappears, but the video resolution becomes - 9184x6888. Also without binning.

Dear, dear friend, if you solve this little bug, I will be very grateful to you)

P.S. Still, what is the maximum resolution of RAW video shooting for Asus ZenFone 7 Pro - 4576x3432 or 9 184x6 888?))

Follow the links below for screenshots illustrating my words, as well as the appearance of the Asus ZenFone 7 Pro rotary camera:

[image]
[image]
[image]

mirsadm, 2021-12-30:
I suspect that capturing at 64 MP is probably a bit excessive! What version of Android does your phone run? I think it's possible to configure the binning options in newer android versions

lordsvarogofmax, 2021-12-30:
I have Android 11. 
Naturally, I have no need to shoot in such a huge resolution - 9 184x6 888))
I just wish that the front-facing camera mode also worked adequately. 
There is no need for binning, the main camera writes video well in a resolution of 3,844x2162 (the closest format to 4K Ultra HD).