mirsadm/motioncam #104: Red line and unsupported cameras pixel 6pro

OverCloker, 2021-12-31:
When using the app, the red bar on the left side of the screen flashes. No access to 4x lens and wide angle. 
[image]
[image]
[image]

mirsadm, 2022-01-03:
Does the red line show up in photos/RAW video?

nocantsin, 2022-01-11:
Chiming in here: no, the red line doesn't appear in the videos. But it's the same issue I reported earlier, that only the main rear camera of Google Pixel phones is recognized/supported by Motion Cam.

LuckyBastard73, 2022-01-19:
Today I asked if somebody could share images from the Pixel 6 Pro on a major German android help forum (because all images I saw form this device hav not really ben the hit).
Two guys there told me that the app crashes on their devices (8.0.6 and 8.0.7 latest beta).
Only one module sees to work (main camera?) and one got a crash when returning from the settings.

https://www.android-hilfe.de/forum/google-pixel-6-pro.4003/sample-bilder-von-der-motioncam.995690.html#post-12889947

mirsadm, 2022-01-19:
yes there are some issues with the Pixel 6. The phone itself seem very buggy at the moment but I'm not discounting the fact that maybe there is something I can do within the app to fix these issues. Unfortunately with these device specific issues it's hard to fix it unless I can test on the phone itself.