mirsadm/motioncam #105: Motioncam tools windows 7 64bit  error with dlls

2blackbar, 2021-12-31:
Hi , i tried to run motioncam tools but i get this error :

The procedure entry point NtQueryWnfStateData could not be located in the dynamic link library ntdll.dll

At first i got error with dxgi.dll , so i downloaded it and copied to folder where motioncamtools is, then i got this error with ntdll, copying it wont help.

Can we get additional converter exe file that would work with command line and wouldnt require libraries from win10 ?
Kinda drag and drop for conainer and it would convert everything to dng
This might be win7 error only, We tried it on 2 pcs with win7 and its the same issue.

mirsadm, 2021-12-31:
Ah yeah it is compiled for Windows 10. It probably would work on windows 7, just need to see if I can compile it targeting it

Gooner447, 2022-01-02:
Hi,Will there be motion tools for mac users ?

mirsadm, 2022-01-02:
Yep, I use a Mac I just haven't had the time to create a package for it yet. I'll try to do both next time I update them.

Gooner447, 2022-01-02:
Thanks, look forwards to it

mirsadm, 2022-01-04:
@Gooner447 Mac build tools are up
https://github.com/mirsadm/motioncam/releases/download/8.0.6-beta4/motioncam-tools-mac-0.0.3.dmg

Gooner447, 2022-01-04:
HI, thanks very much, I will try it out straight away.  The audio syncing
problem seems to be fixed and I am able to sync up audio and video.  I have
tried the higher frame rates on my S20+ but without much success so far,
60fps crashes the app and 120 fps records at 30 fps, as you probably know
the S20+ native app is cappable of these speeds but I'm sure you will all
sort it out.
On a side note, I am retired and this app is giving me hours of
entertainment, Thanks

mirsadm, 2022-01-04:
@Gooner447 that's great to hear. Unfortunately there's not much I can do about the frame rate issues. There are simply too many phones out there and it's just me working on this as a hobby

@2blackbar I'm going to close this for now, I think Win10 is going to be the minimum for the time being until I can figure out how to target something older

closed 2022-01-04 by mirsadm