mirsadm/motioncam #106: Samsung S21 (Exynos 2100) video stutters.

dalcantom, 2022-01-02:
Ok, just tried with my new Samsung S21 (Exynos 2100) and write speed is ok but final video stutters. There's no stutter on Samsung Galaxy Zflip 3 (SnapDragon, it crashes after few seconds but it's a know bug and until crash the video is perfect). On the preview the video already lags, it's not smooth as should be.

The reason could be that the frame rate is inconsistent. Regardless the frame rate set, the actual FPS is not stable and for sure it's not the same of the settings.

mirsadm, 2022-01-02:
if the preview is not smooth it sounds like it's dropping frames already. You could try changing to RAW16 in the settings which may help. Does it stutter before the memory bar has turned red and become full?

dalcantom, 2022-01-02:
Memory and disk write are green, at low levels and never become full. With setting 30fps the real frame rate is about 24-25 and not stable. With setting  25fps it reaches 20-21. RAW16 is not available on my S21 Exynos, I've seen that on Qualcomm ZFlip 3 of my wife is available.

mirsadm, 2022-01-02:
What's the output resolution?

dalcantom, 2022-01-02:
4032 x 3024

mirsadm, 2022-01-03:
https://github.com/mirsadm/motioncam/releases/download/8.0.6-beta4/MotionCam.8.0.6-beta4.apk

Few changes here that may help

bornd3vil, 2022-01-05:
unfortunately no improvement for now

[image]

bornd3vil, 2022-01-18:
in fact, on my Tablet Samsung S5e the preview is smooth, when I change fps in settings the change is immediately visible! the OIS works well and it reaches 60FPS, of course when I start to record memory became full in a while.

in S21 the preview is laggy not smooth, the OIS desn't work, when I change fps in settings the effect is not visible, I need to exit and re-enter in app, however 30 is the max, but when I record the fps is not 30, and the video stutters as the preview.

it's easy to see that if preview is not smooth the video cannot be different, so I hope I can help you to make it work as S21 is a great phone, and I'm tired to bring with me all those mirrorless devices to have a video RAW for my jobs..

I'm ready to pay to make it work, if possible as video RAW is not a joke, it's the solution for my back diseases! :)