mirsadm/motioncam #107: [Request] Matrix group for general discussion

epicfilmmaker93, 2022-01-02:
While Github works great for issues and pull requests, I think it would be great to have a Matrix group for general discussion. I've seen there's already a Russian Telegram group, but I think there should be an official one for English speaking users

mirsadm, 2022-01-02:
Yes that is a good idea. I was thinking of setting up a Discord server

epicfilmmaker93, 2022-01-02:
I would ask you to consider an open source platform instead - https://drewdevault.com/2021/12/28/Dont-use-Discord-for-FOSS.html

experimentator1709, 2022-01-02:
Great idea if the group is open. How will you do it - give a link - I will post it in the Russian-language telegram group.
https://t.me/MotionCam

FilipZamec, 2022-01-03:
Yes, discussion app would be great 👍.
On mi 11U latest 8.0.6 version works for all lenses and 60 fps on them too. Thank you 🙂.
When you introduced compression. File sizes are aprox. 2.5 smaller. So I can shoot much more stuff.
Small bugs:
After shooting 60 fps I usually have to restart the app or it crashes.
And if I choose 120 fps, which is not supported on my phone, app crashes on start, so I have to uninstall and install again.
Otherwise dream come true. . 🙏

mirsadm, 2022-01-04:
Will be creating one soon. I think probably with Slack. FOSS would be good but it's not convenient and I don't think many people would want to bother installing things they're not already familiar with.

closed 2022-01-04 by mirsadm