mirsadm/motioncam #108: Problem when recording raw video in version 8.0.6 (any frame rate)

He5am2099, 2022-01-03:
Hello.
Before addressing the issue, I wanted to thank you for creating this wonderful app.
Issue:
After updating to the latest version of the app (8.0.6) , my device (galaxy note 10 plus) can't record raw video at any frame rate (even supported frame-rates such as 24fps....). I will happily give or send any information that can help in troubleshooting this problem. 
Thanks again for putting the time and effort into developing this app.

mirsadm, 2022-01-03:
When you say it doesn't record at any FPS, could you elaborate?

He5am2099, 2022-01-03:
Yes, when I press the record button, the app starts recording for a brief second, but then automatically stops, a "saving video" card appears on the middle of the screen and when I check the recorded files with this version of the app, they don't have any frames, fps or Metadata.... attached to them:
[image]

mirsadm, 2022-01-03:
If you click the little arrow are there any frame rates listed under supported frame rates?

He5am2099, 2022-01-03:
Yes, in fact, just like the previous version, frame rates such as 15, 20, 24 and 30 fps are listed as supported frame rates.

mirsadm, 2022-01-03:
Are you able to get me ADB logs when you run the app and click record?

mirsadm, 2022-01-03:
https://github.com/mirsadm/motioncam/releases/download/8.0.6-beta2/MotionCam.8.0.6-beta2.apk

Have a go of that one and let me know if it changes anything. If not, I'll need logs from your device to investigate further

He5am2099, 2022-01-03:
Sorry for taking so long to reply. I was figuring out how to get a proper adb log file of the app. I'll try the version that you sent to me and will send an adb log if the problem was not fixed. Thanks a lot.

He5am2099, 2022-01-03:
So, the issue is still apparent. I'll get an adb log of the app and send it to you. Thanks again 🙏.

mirsadm, 2022-01-03:
https://github.com/mirsadm/motioncam/releases/download/8.0.6-beta2/MotionCam.8.0.6-beta3.apk

One last attempt before I need the logs :D

He5am2099, 2022-01-03:
So the newer version didn't work either. I will send the logs now.

mirsadm, 2022-01-04:
Is this this still an issue for you?

He5am2099, 2022-01-05:
Unfortunately, the issue is still there. I'm really sorry for taking so long to send the log file. I used logcat extreme and many other apps to record motion cam's log, but failed. Do you have any suggestions for apps that I can use to record the log? Or should I use adb?

He5am2099, 2022-01-05:
There's about 6GB of free space on my device.

He5am2099, 2022-01-05:
Because most of them either don't give the specific app's logs or don't work.

mirsadm, 2022-01-05:
I'm fairly confident that's the issue so I'll put out another beta today and that should fix it

He5am2099, 2022-01-05:
Ok, thanks. I'll clean up my phoen a bit to have more storage.

He5am2099, 2022-01-05:
The problemwith the recording ability is fixed. (It was my bad). Thanks again for helping. 
I think that another problem has happened. I recorded a few clips (where the recording function didn't work) and those corrupt files are causing the app to crash when I open the "manage videos" section of the app. Can you suggest an app or ... that I can use to record the problem's log?

mirsadm, 2022-01-05:
Are you able to send me the recorded container files by any chance?

He5am2099, 2022-01-06:
Again, I'm really sorry, but I hadn't set the container files' folder to a custom one after deleting and reinstalling the app, so I think that they're in the app's folder (which I couldn't access because my device's not rooted).

He5am2099, 2022-01-06:
Never-ending, that issue got solved too. I deleted the files just before the app could completely load them and crash and it worked. Sorry for taking so much of your time. And thank you for helping. 🙏

closed 2022-01-06 by He5am2099